Blueforcer/awtrix3 #417: Auto Bright

DennisXK, 2023-11-28:
Hi, ich habe mir eine zweite Ulanzi TC001 zugelegt und mit der Firmware 0.9 bespielt. Ich musste feststellen, dass wenn ich die Helligkeit auf Auto stelle, diese viel heller ist als bei meiner ersten Ulanzi mit der Firmware 0.88.

Liegt es an der Firmware, oder hat Ulanzi einen andere LDR verbaut?

Blueforcer, 2023-11-28:
Du kannst die grenzen nach deinen Wünschen per dev.json einstellen:
https://blueforcer.github.io/awtrix-light/#/dev

salabim, 2023-11-28:
Wie kann ich auf die dev.json-Datei auf meinem Ulanzi-Gerät zugreifen/schreiben?
Ich arbeite mit Windows.

Blueforcer, 2023-11-28:
Über den Filemanager auf den du per Webinterface kommst.

DennisXK, 2023-11-28:
Ok, ja das klappt löst aber das Problem nicht wirklich. Das Regelverhalten wird einfach nicht komplett ausgenutz.

default ist 180, ist das das technische maxium?

Blueforcer, 2023-11-28:
255 ist max

DennisXK, 2023-11-28:
Bei der 0.88 ist es besser, halte ich den Finger auf die LDR zeigt er mir in den Einstellungen 2% was einem Wert von 3,6 entspricht, wenn min und max 0-180 ist. Haltet man eine Taschenlampe dagegen kommt er bei der 0.88 sowie bei der 0.9 leider nicht auf das Maximum. Also man bekommt die maximale Helligkeit nur mit der manuellen Einstellung auf 100%.

evt.l kann das jemand prüfen, nicht das es nur bei meinem beiden Ulanzi's der fall ist.

Blueforcer, 2023-11-28:
Die Berechnung wurde gefixed, damit diese auch den richtigen Werten entspricht.
Das es vorher dunkler war, war der eigentliche Fehler.

https://github.com/Blueforcer/awtrix-light/pull/340

DennisXK, 2023-11-29:
Ok, dann ist das halt so, trotzdem erricht die Matrix nicht ihren Anfang und Endpunkt beim abdunkeln bzw. anleuchten des Seonsor. 

Kann man selbst überprüfen Sensor abduckeln und in das Bright Menü gehen. Dann von Auto auf Man umstellen. Da sieht man wieviel Prozent die Matrix hat.

Blueforcer, 2023-11-29:
Das Menü nach dem umschalten ist nicht aussagekräftig, da dort wieder eine Umrechnung stattfindet.
Ich habe die Rohwerte geprüft und die Berechnung passt.

Zuhalten = raw 0 = 0% = min_brightness

Taschenlampe = raw 1023 = 100% = max_brightness.

DennisXK, 2023-11-29:
ok, finde es leider in einem dunkeln Raum leider noch zu hell. Eine Empfindlichkeitprofil Einstellung wäre cool.

Blueforcer, 2023-11-29:
Wenn du 1 als min_brightness angibst und den LDR zuhählst geht er mit der Helligkeit bis auf eins. Noch eins weiter und die Matrix wäre aus. Mehr geht nicht. Dazwischen gibt's nichts.

olanwe, 2023-11-29:
Ich habe mit der Helligkeitseinstellung auch ein bisschen Probleme. 
Im Moment habe ich im Wohnzimmer "angenehme Abendbeleuchtung". Die Uhr ist aber bereits auf Minimum (d.h. viel zu dunkel), wobei der Sensor in "stats" folgende Werte zeigt:

"lux": 0, "ldr_raw": 211, "bri": 2

Interessant ist, dass der Lux-Wert hier bereits auf 0 steht. Decke ich den Sensor mit Papier ab, bekomme ich: 

"lux": 0, "ldr_raw": 12, "bri": 2

Ich kann zwar mit min_brightness die Helligkeit hochsetzen, aber Nachts (also bei kompletter Dunkelheit) bleibt die Uhr auf dem höheren Level, während es da ja gerne auch bis zum Anschlag runter gehen darf. Ich denke auch, dass hier ein Faktor (evtl. mit Offset) helfen würde. Die Uhr steht bei mir frei auf einer höheren Kommode, aber nicht in unmittelbarer Nähe einer Lichtquelle.
(Ich habe den Eindruck, dass es bei der 0.90 noch etwas besser gepasst hat und es mit 0.91 schlechter geworden ist - kann mich aber täuschen.)

olanwe, 2023-11-30:
Sorry... 😊Ich denke eines der Hauptprobleme ist, dass der Sensor nach oben zeigt und nicht nach vorne. Stellt jemand die Uhr in ein dunkles Regal oder in eine dunkle Ecke, dann passt der Messwert natürlich nicht und ist immer viel zu dunkel. Steht die Uhr unter einer Lampe, ist sie zu hell. Ich habe mir echt schon überlegt, einen kleinen Spiegel hinzukleben - wenn es nicht so doof aussehen würde... 😂

DennisXK, 2023-11-30:
habe die Version 0.91 aufgespielt, mit dieser Version läuft es deutlich besser im vergleich zur 0.9 und 0.88.

Um es jedem gerecht zumachen, könnte man doch einem Multiplikator einfügen, der mit dem Rohwert verrechent wird!?

luebbe, 2023-12-11:
Vermutlich ist der Helligkeitsverlauf obendrein nichtlinear und je nach Charge der in der Ulanzi verbauten LEDs auch noch unterschiedlich.

Blueforcer, 2023-12-23:
https://github.com/Blueforcer/awtrix-light/releases/tag/0.92

Mehr kann ich leider nicht mehr tun :)
Wenn wer noch ideen hat, darf gerne einen PR aufmachen

closed 2023-12-23 by Blueforcer

luebbe, 2023-12-23:
Für mich schaut es gut aus. Bei niedrigen Helligkeiten ist es immer sehr schwierig weiche Übergänge hinzubekommen.

Blueforcer, 2023-12-23:
> Für mich schaut es gut aus. Bei niedrigen Helligkeiten ist es immer sehr schwierig weiche Übergänge hinzubekommen.

Ja weil die LEDs zu spät angehen. Ich denke ohne gleichbleibende Hardware Qualität und Messgerät ist es nicht möglich die ordentlich zu kalibrieren.