Blueforcer/awtrix3 #418: [QUESTION] Is it possible to use 2 devices with Home Assistant

istornz, 2023-11-29:
Hi @Blueforcer,

Thanks for your awesome work!
Sorry if I post a question here :/

I'm using HomeAssistant & I have 2 Ulanzi with Awtrix Light flashed on it. How I can connect these 2 devices with my HA instance?
I connected one but the second is not possible because Mqtt broker doesn't support more than 1 device.

Thanks for the help in advance!

distanzcheck, 2023-11-29:
https://youtu.be/N0NKPJzGHuA?si=Ughtp-5z9G4wO6FO

istornz, 2023-11-29:
> https://youtu.be/N0NKPJzGHuA?si=Ughtp-5z9G4wO6FO

Thanks, I already followed this tutorial to connect the first device but you can't add another using HA Mqtt broker ...

distanzcheck, 2023-11-29:
I have included a total of three ulanzi clocks in this tutorial.

istornz, 2023-11-29:
Oh nice working well with 0.91!

I tried with 0.90 version but UI of clock freezes...

Thanks a lot 😁

closed 2023-11-29 by istornz