Blueforcer/awtrix3 #419: [FEATURE REQUEST] Allow to adjust volume for DFPlayer

Lutty76, 2023-11-30:
# Feature Request

### Is your feature request related to a problem?

My problem is that I cannot adjust volume

### Describe the solution / feature you'd like

In global.cpp
you have this 
```
#ifndef ULANZI
    Settings.putUInt("VOL", VOLUME_PERCENT);
```
and this

```
#ifndef ULANZI
    VOLUME_PERCENT = Settings.getUInt("VOL", 50);
    VOLUME = map(VOLUME_PERCENT, 0, 100, 0, 30);
```

but we can't set this setting for api as this variable doesn't exist in DisplayManager.cpp
so you just put default value (50) in setting and reload it

### Describe alternatives you've considered

Just add : 

`  VOLUME_PERCENT = doc.containsKey("VOL") ? doc["VOL"].as<int>() : VOLUME_PERCENT ;`
in  DisplayManager_::setNewSettings
for allow use to change volume

### Additional context

Add any other context or screenshots about the feature request here.

Blueforcer, 2023-11-30:
Currently you can set the volume via onscreen menu

Lutty76, 2023-11-30:
I don't have plugged the screen for the moment 😅 so i didn't see this thanks for information. But can be done by API should be a good thing, no ?

Blueforcer, 2023-12-23:
https://github.com/Blueforcer/awtrix-light/releases/tag/0.92

closed 2023-12-23 by Blueforcer