Blueforcer/awtrix3 #420: Rebooting after setting MQTT settings

JeGr, 2023-12-01:
# Bug report

## Describe the bug

I have already read (https://github.com/Blueforcer/awtrix-light/issues/165) but that didn't seem to be my problem? Anyway as per my title I have the problem of a reboot loop every time I enter the MQTT settings. I can configure every part of AWTRIX but as soon as I touch MQTT, the device is gone, doesn't get anywhere near getting configurable again (no HTTP available anymore/not responding).
Newest 0.9 firmware flashed. MQTT Broker is on the same network as the device (a separate VLAN where IoT stuff is housed) and is a simple mosquitto instance that was set up without much configuration at all. Simple setup, configured user/passwords to test. I already connected several Tasmota devices to test with the broker which do just fine and pop up as soon as I save the settings.

But as soon as I enter the MQTT host, user, pass in AWTRIX, the device reboots and after checking my DHCP server logs, I see it reconnect/getting an IP around every 10s so the device doesn't even get in a usable state anymore where I could reach the HTTP server.

So I'm open to debugging but at a loss what to provide to get it to talk to my broker.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.9
  - Target: Mosquitto broker v2.0.13 (running on FreeBSD)

## To Reproduce

Steps to reproduce the behavior:

 - Flashed AWTRIX light 0.9
 - minimal setup
 - device is working
 - enter MQTT credentials
 - device enters reboot-loop which can be watched in DHCP logs by retrieving DHCPrequests every 10s
 - no web server available

## Expected behavior

 - login into MQTT broker
 - boot to completion
 - reachability via HTTP

Happy to provide any other data I've available, mosquitto is just freshly installed, no special configuration besides pwfile.

Blueforcer, 2023-12-01:
Check serial console, and check your mqtt messages via mqtt explorer. 
I'm pretty sure you're sending a reboot command.

JeGr, 2023-12-01:
Thanks for the response!

Sorry, my bad, it seems it isn't a reboot loop, the device I identified getting a DHCPoffer every few seconds wasn't the MAC of the TC001 but a very similar one. Should have checked better.

The problem seems to be in the toggle of "Homeassistant Discovery". If I save MQTT credentials without that toggle and reboot, the device comes up fine and I see it responding in MQTT explorer.
If I toggle the discovery to on and reboot, the device seems to hang with MQTT... on screen doing nothing. Also the HTTP server is not responding.
After more then 5min it suddenly flips to the date/time view with the calendar icon but doesn't respond to any input (no buttons respond, HTTP still not available). 

The only thing that is responding is the IP via ping, but nothing else. Also I doesn't seem to get any console output out of the device. Perhaps wrong settings? Haven't found any that produce any output at all.

Blueforcer, 2024-03-08:
DHCP has nothing to to with homeassistant discovery.
Does you problem still exist with 0.94 or can we close this issue?

closed 2024-03-17 by Blueforcer