Blueforcer/awtrix3 #423: [VIDEO TUTORIAL THAT EXPLAINS HOW TO ADD AT LEAST ONE FLOW TO NODE RED]

sysadmin-info, 2023-12-03:
How to implement this? [youtube follower count](https://flows.blueforcer.de/flow/erM3xABvA8Lz). I have already installed Node Red in Home Assistant, uploaded the flow into the Node Red, then provided the API key and channel ID. Saved. Deployed and nothing is displayed. It says about setting up AWTRIX MQTT prefix, but there is no field called prefix, so I need an explanation where to put the prefix. Can someone explain this, please?

Blueforcer, 2023-12-03:
Did you read the Documentation?
https://blueforcer.github.io/awtrix-light/#/apps

sysadmin-info, 2023-12-03:
> Did you read the Documentation?
> https://blueforcer.github.io/awtrix-light/#/apps
> 
> P.s, UPPERCASE will not prioritize your issue;)

I did it on my own exactly the way it is described but when I deploy it to ulanzi it does not show on the Ulanzi yc001 screen. , I flashed Ulanzi and MQTT is working in Home Assistant, but for some reason it does not work through Node Red that is also installed inside Home Assistant. That is why it would be good to see the video and how someone is doing it. What is funny, when I deployed different workflow through n8n it worked like a charm. Any explanation why Node Red does not display anything on Ulanzi screen? Red Node shows that whole workflow works correct Andis able to connect to MQTT broker installed via Mosquito in Home Assistant. Is there any additional step that was not described and is crucial? I am using the newest Home Assistant Supervised and all add-ons are up to date.

sysadmin-info, 2023-12-05:
I finally solved it. Tony's flow is using Ulanzi's IP address. Blueforcer's flow is using MQTT broker IP address and MQTT user that can be defined inside the Home Assistant instead inside Mosquito MQTT Broker addon configuration section. UI recorded whole series about this device and how to configure everything step by step. I will but it on my YouTube channel, when I will edit videos so they will be available on December 6, 2023.

sysadmin-info, 2023-12-09:
Here you have a series of videos about it, so the request has been done by me: https://sysadmin.info.pl/en/series/ulanzi-tc001-smart-pixel-clock/

closed 2023-12-18 by Blueforcer