Blueforcer/awtrix3 #424: [BUG] auto brightness in 0.91 is dimmer

golles, 2023-12-04:
First and most importantly, thank you for making and sharing this project, I appreciate your work and dedication!

# Bug report

## Describe the bug

Recently the `0.91` release came out and I installed this on my 2 clocks.
I immediately noticed the screen was dimmer than before.
I've changed one clock back to version `0.90` to see at the same location what the effect is.
I took a picture (see below) of both clocks in the same place with some daylight coming in (the roller curtain only open a little bit), as with less light it is more visible (like with the current winter weather conditions).

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.91

## To Reproduce

Steps to reproduce the behavior:
Install version `0.91` after running version `0.91`.

## Expected behavior

I would prefer to have the auto brightness to be how it was in the previous version, as when it's darker (outside) its more difficult to read the screen of the clock.

## Screenshots

[image]
Left is the clock running version `0.90` and right is the clock running version `0.91`

## Logs

N/A

## Additional context

N/A

Blueforcer, 2023-12-08:
set the min_brightness via dev.json higher to fit your needs.

closed 2023-12-08 by Blueforcer

golles, 2023-12-08:
@Blueforcer what would be an appropriate value to make it comparable to version 0.90?

Blueforcer, 2023-12-08:
Default is 2. 
Try 3 or 4

golles, 2023-12-09:
Setting `min_brightness` is a nice solution during the night, but wouldn't increase the brightness during the day

Blueforcer, 2023-12-09:
Too dark, too bright 😅
Play also with the max value

golles, 2023-12-09:
That's not what I was trying to say. In general, the clock is too dim since the latest update.
Setting `min_brightness` is only effective when it's dark, as it would now allow the backlight to go less than the set value. During the day, when it's less dark in the room, the `min_brightness` has no effect, as the auto brightness stays higher. In my previous photo, you can see the difference during winter daytime. The right clock is a lot dimmer than the right, but still not near the `min_brightness`

Blueforcer, 2023-12-09:
In the previous version it was a bug, and the LDR reading was not correct so  the matrix wasn't able to detect bright environment and it was too dim.

Now it's correct. There is no other option, sorry.

golles, 2023-12-10:
Ok, thanks for the support