Blueforcer/awtrix3 #427: [BUG] Base64 icons not working without "=" padding

magcode, 2023-12-10:
# Bug report

## Describe the bug

Some base64 encoded icons are not recognized.

Example is attached 
[image]

base64 of this image is 

```
/9j/4AAQSkZJRgABAQAAAQABAAD/2wBDAAgGBgcGBQgHBwcJCQgKDBQNDAsLDBkSEw8UHRofHh0aHBwgJC4nICIsIxwcKDcpLDAxNDQ0Hyc5PTgyPC4zNDL/2wBDAQkJCQwLDBgNDRgyIRwhMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjL/wAARCAAIAAgDASIAAhEBAxEB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQAAAF9AQIDAAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3ODk6Q0RFRkdISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWmp6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEAAwEBAQEBAQEBAQAAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSExBhJBUQdhcRMiMoEIFEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElKU1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3uLm6wsPExcbHyMnK0tPU1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwD0TW9XstQtmMVzNcW8Esbzx22VZADwc8ZHQ4zRRRXApO1z0YaKyP/Z
```

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.91

## To Reproduce

Send above base64 as icon.

## Expected behavior

All base64 encoded icons work.

Blueforcer, 2023-12-10:
Currently I don't know how to detect a base64 string except of the trailing =.

Maybe according to a length

magcode, 2023-12-10:
Regex maybe?

https://stackoverflow.com/a/8571649

magcode, 2023-12-10:
Yeah, length would be much easier. Everything above 32 (?) chars can be safely treated as a base64.

fasterit, 2023-12-12:
As I wrote on Discord:
JPGs will always start with hex FF D8 FF which will lead to the base64 start of /9j/ (including the /'s). So if you use that to detect, you are safer. Also GIFs start with GIF8, so you can use R0lG at the start of base64 to detect these.
(you can combine this with the length test, in case somebody wants to make a very strange icon file name :smiley:)
/Greets DLange

Blueforcer, 2023-12-23:
Checking the jpg header would require to encode the base64 string first. GIFs as base64 are still not possible.
For now awtrix checks the stringlenght.

https://github.com/Blueforcer/awtrix-light/releases/tag/0.92

closed 2023-12-23 by Blueforcer