Blueforcer/awtrix3 #429: [BUG] Kalendersymbol wird nicht ausgeblendet

AngieSoft, 2023-12-13:
Seit update von 0.7 auf 0.91 wird auf dem onscreen-Display das Kalendersymbol in der Zeitansicht nicht mehr ausgeblendet.

[image]

Blueforcer, 2023-12-18:
War die neue Zeitansicht in der 0.7 überhaupt schon drin ?^^
Du musst den Time-Mode (TMODE) per settings API ändern.

AngieSoft, 2023-12-18:
Ja, da konnte man den Kalender per Tasten am Gerät ausblenden, so dass nur noich die Zeit, inkl. Sekunden angezeigt wurde.

Blueforcer, 2023-12-18:
> Ja, da konnte man den Kalender per Tasten am Gerät ausblenden, so dass nur noich die Zeit, inkl. Sekunden angezeigt wurde.

Nein, du konntest lediglich die Kalender App deaktivieren. Das geht weiterhin.
Das Kalender Icon gehört zur Time-App seit Version "keine ahnung" xD
Du kannst sie per settings API nach belieben anpassen

closed 2023-12-23 by Blueforcer