Blueforcer/awtrix3 #430: Unable to control brightness via MQTT

DaveBKC, 2023-12-14:
Hi, not a bug, I assume. I trying to understand what I'm doing wrong.
I'm able to successfully send commands via MQTT I get the expected results. A couple of examples of what works for me:
```
awtrix_b91f90/moodlight
{"brightness":70, "color":[155,38,182]}
```
```
awtrix_b91f90/indicator1
{"color":[0,255,123]}
```

My issue is that sending a brightness command, regardless of the payload, turns the matrix black. The only way I know to turn it back on is via the app. Here's what I'm sending:

```
awtrix_b91f90/settings
{"BRI":[10]}   
```

Are settings not supported yet? 
My use case if it helps: My clock is mounted near a wall where auto-brightness doesn't work. I was planning to create an HA automation to control the brightness through sunrise/sunset. 

Sorry if this is in the wrong place. Thanks for reading.

Blueforcer, 2023-12-14:
Don't use the brackets [10] around the 10. Also make sure automatic brightness control ist deactivated.

DaveBKC, 2023-12-14:
I can't I didn't already try that. Thanks!

Gerrit-KMeier, 2023-12-24:
I just tried it with V0.93 using this command
` mosquitto_pub -t awt/settings -m '{"BRI":200,"ABRI":false}'`

works like expected!
Note: My mqtt prefix ist just `awt`

Maybe it's also worth to mention, that a dark font-color stays quite dark - even with full brightness. So consider changing the foreground color, accordingly
` mosquitto_pub -t awt/settings -m '{"TCOL":"#FFFFFF","TIME_COL":0}'`

DaveBKC, 2023-12-25:
Yeah, it was the brackets. I got it sorted a while back. Thanks though.

closed 2023-12-25 by Blueforcer