Blueforcer/awtrix3 #431: [FEATURE REQUEST] Animation icon Database

MasterVigori, 2023-12-14:
Hello,
is it possible to use the old animation icon database from Awtrix 2.0? 
[image]

Blueforcer, 2023-12-14:
It's not possible, both are different formats.

next version of the official awtrix light app will contain a 32x8 Database

Blueforcer, 2023-12-14:
The only way is you can download the GIFs from the awtrix 2 Webinterface and resize it to 32x8 and upload it to awtrix light.

MasterVigori, 2023-12-14:
Ok, thanks I will try that and upload it to Awtrix light

Thank you

MasterVigori, 2023-12-14:
> The only way is you can download the GIFs from the awtrix 2 Webinterface and resize it to 32x8 and upload it to awtrix light.

Do you have any idea which program I can use to convert the tetris.bin file to tetris.gif?

[Animation DB](https://awtrix.blueforcer.de/animations/ )

closed 2024-01-28 by Blueforcer