Blueforcer/awtrix3 #434: [BUG] app does not get marked as staled if no other app is active

matthias-bb, 2023-12-18:
# Bug report

## Describe the bug

When I start a custom app using "lifetimeMode=1" (app gets marked as staled after defined lifetime) it only gets marked as staled if an app switch occurs (more than one app is active and auto transition is enabled or app switch is forced by pressing left or right key).

## Additional information

- Devices involved:
ul  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.91

## To Reproduce

Disable the native apps and start a custom app with lifetime=10 and lifetimeMode=1.

## Expected behavior

After the given lifetime of 10 seconds I expect the app to get marked as staled.

## Additional context

I am currently using the clock to monitor that my Raspberry Pi is up & running. Therefore I use a cron job which periodically updates a "pi alive" custom app via http. If this is the only app being displayed it never turns to staled even if my Raspberry is not alive anymore.

Blueforcer, 2023-12-18:
Thats not a bug, Lifetime is only checked during app switches to avoid stuttering while the app is active, otherwise i would need to check it every second

closed 2023-12-23 by Blueforcer

Clem-, 2024-05-12:
That would have been nice to be able to delete a single app in this manner.
I upvote for this feature request.