Blueforcer/awtrix3 #435: [QUESTION] esp32 with matrix panel

PaulBiod, 2023-12-19:
Hi,
I'm not using the ulanzi device, I use an esp32 connected to a 32x8 panel.
I followed de process; flashed (with [Ulanzi TC001 and custom builds flasher]) , opened webui, set up wifi, mqtt user. Alse, Home assistant is seing the panel.
But instead of the hour, I have meaningless characters.
I create a dev.json file
[image]
(tried with 0, 1 and 2, almost same result.
The esp32 has pin 32 connected to the data in line of the matrix panel.
Am I doing something wrong ?
Thanks

Blueforcer, 2023-12-19:
That's look like a faulty connection, not a software problem. Check your cables. It's also better to solder it.

PaulBiod, 2023-12-19:
So, pin 32 is correct ? connection about voltage are ok, and the dupont cable from pin 32 is soldered

PaulBiod, 2023-12-19:
Ok, panel is faulty... even with wled and a solid color, it messes completly

closed 2023-12-19 by Blueforcer

Demel75, 2025-03-31:
I got similar problem. I had migradte from esp8266 (awtrix2) and rgb-matrix work flawless. Any idea?