Blueforcer/awtrix3 #437: [BUG] Unable to update

PaulBiod, 2023-12-20:
Hi,
i am using 0.90 
[image]
When I go to "update" tab, I select the bin file 0.91, then click update, after a few moment, the page shows as follow : 
[image]
And when I refresh the gui, it's still in 0.90

Blueforcer, 2023-12-20:
Try the update function in the buttons Menü, or flash it via USB

apachelance, 2023-12-22:
Not sure if this is related. However I had issues with timeouts as well, not only using the update function but also regarding general connectivity. Could be a network issues. However I went back to 0.89.

Blueforcer, 2023-12-23:
Just tested it with 0.92. works as it should. Also tried updating via app or onscreen menu.

Blueforcer, 2023-12-28:
@PaulBiod Can this issue be closed?

PaulBiod, 2023-12-29:
Hi @Blueforcer 
sure, it works perfectly. Thanks

closed 2023-12-29 by PaulBiod