Blueforcer/awtrix3 #438: [FEATURE REQUEST] AWTRIX should maybe use a different font

RobertoCravallo, 2023-12-22:
# Different fonts for AWTRIX

### Is your feature request related to a problem?

NO!

### Describe the solution / feature you'd like

Upload own fonts (seems to not be possible as per a previous request!!)

### Describe alternatives you've considered

Use AWTRIX as is!

### Additional context

Basically, this thing just rocks!!! Thank you, by the way!!! The "problem" is with lower case letters like the "w", which are hardy recognizable. Maybe you can generally use a "cleaner" font.

On the AWTRIX 2 this was possible and I saw a video with a really nice "w". ;-)

[image]

Blueforcer, 2023-12-22:
Duplicate of https://github.com/Blueforcer/awtrix-light/issues/53

Awtrix 2 uses the same font, and it also wasn't possible to change the font.

RobertoCravallo, 2023-12-22:
Ups, misunderstood something there!!! Maybe it was a bigger display? Thanx.

closed 2023-12-23 by Blueforcer