Blueforcer/awtrix3 #441: Turn off christmas Let it snow

hytrek, 2023-12-23:
Hi there,
I just installed the latest Firmware and yes, it was mentioned I can disable the Snowfall, but how? I am using you iphone app. cannot find a setting for the dev.... Please let me know how to...
Many Thanks and merry christmas

Blueforcer, 2023-12-23:
https://blueforcer.github.io/awtrix-light/#/dev

hytrek, 2023-12-23:
I understand... But all I have is either the app or the ioBroker plugin... I have no developer setup or so.... So honestly I have no clue what to do... I like the effect for the moment but christmas and winter will be over one day :-)

theunknownstarman, 2023-12-23:
how is the correct syntax for disabling?

Blueforcer, 2023-12-23:
1. Go into your AWTRIX filmanager (**http://[AWTRIX-IP]/edit**)
2. Create a **dev.json**
3. Paste **{"let_it_snow":false}**
4. Save the file
5. Reboot awtrix

cobra98, 2023-12-23:
In other words create a file with a notepad on windows or vi/nano on linux with file name "dev.json and upload it to http://AWTRIXIP/edit and reboot
Maybe a way to turn off it in the awtrix light app on android would be a better way.

[image]

should look like this

Blueforcer, 2023-12-23:
You can create and edit files directly in the Filemanager.

Read the documentation for further dev settings.

https://blueforcer.github.io/awtrix-light/#/dev

marcgauch, 2023-12-24:
Pain. I use the light in my bedroom. The snow is too bright and I hate animations. Disabling it was way too much work. Like OP I only use the app. Not amused. Just wanted to sleep.

GreenTentacle, 2023-12-24:
I added a `dev.json` and I still have the snow falling over every app.  

Honestly, this is something that I'd like to have easily configurable -- it changes what the end user is expecting to see, without asking their permission or giving them an easy way out.

GreenTentacle, 2023-12-24:
> I added a `dev.json` and I still have the snow falling over every app.
> 
> Honestly, this is something that I'd like to have easily configurable -- it changes what the end user is expecting to see, without asking their permission or giving them an easy way out.

Whoops, this was a stray character in my json... disabled now

distanzcheck, 2023-12-24:
[dev.json](https://github.com/Blueforcer/awtrix-light/files/13760908/dev.json)

Nimm mal diese dev.json und starte danach die neu.

Take this dev.json and then restart it.

sansonifabio, 2023-12-24:
@Blueforcer ich finde es mit dem Schneefall sehr schön, musste aber auch erst vertehen was eine dev.json ist. 
Ich arbeite fast nur mit deiner App und natürlich mit ioBroker und dem Awtrix Adapter. 
Vieleicht hats du ja mal Zeit den Schnnefall in die App zu konfigurieren dann kann man sich das zu Weihnachten selbst aus und anschalten. ;-)

**Frohe Weihnachten und vielen vielen Dank das du Awtrix-light machst.**
Grüße
Fabio

Blueforcer, 2023-12-24:
> @Blueforcer 
> ich finde es mit dem Schneefall sehr schön, musste aber auch erst verstehen was eine dev.json ist. 
> Ich arbeite fast nur mit deiner App und natürlich mit ioBroker und dem Awtrix Adapter. 
> Vieleicht hast du ja mal Zeit den Schnnefall in die App zu konfigurieren, dann kann man sich das zu Weihnachten selbst aus und anschalten. ;-)
> 
> **Frohe Weihnachten und vielen vielen Dank das du Awtrix-light machst.**
> Grüße
> Fabio

Die App kann nicht in die Dev schreiben.
Sieht eher so aus als müsste ich heute nochmal ein Update nachschieben in dem es wieder ausgeschaltet wird.
 
Es ist erstaunlich wieviele Leute keine Ahnung haben was 
1. Awtrix alles kann
2. Eine JSON ist
3. Die Dokumentation ist.

Und dennoch irgendwie awtrix nutzen 😄

sansonifabio, 2023-12-24:
Es ist wie es ist 😁😉

Blueforcer, 2023-12-24:
https://github.com/Blueforcer/awtrix-light/releases/tag/0.93

closed 2023-12-24 by Blueforcer

sansonifabio, 2023-12-24:
> https://github.com/Blueforcer/awtrix-light/releases/tag/0.93

... ich danke dir ganz herzlich und es tut mir wirklich echt leid das User stellenweise so harsch reagiert haben.

smarthomejunkie, 2023-12-24:
Thank you Stephan for all your hard work. I appreciate all the efforts that you are making. Don't be sad about the harsh comments of some people. They probably have no idea how much work you put into this awesome free piece of firmware. I wish you a very merry Christmas and a happy 2024!

Blueforcer, 2023-12-24:
> > https://github.com/Blueforcer/awtrix-light/releases/tag/0.93
> 
> ... I thank you very much and I'm really sorry that users reacted so harshly in some places.
> 
> ... ich danke dir ganz herzlich und es tut mir wirklich echt leid das User stellenweise so harsch reagiert haben.

It's not your fault.

I assume a certain basic understanding of this project. It's a shame to get upset about how I program my project.
I have already deleted comments here that were really offensive.

Blueforcer, 2023-12-24:
> ... I thank you very much and I'm really sorry that users reacted so harshly in some places.
> 

It's not your fault.

I assume a certain basic understanding of this project. I think it's a shame to get upset about how I program my project.
I have already deleted comments here that were really offensive.

theunknownstarman, 2023-12-24:
@Blueforcer 

Ich sehe es genauso wie meine Vorposter. Bitte nicht ärgern, es gibt immer irgendwelche Nörgler die nicht im Ansatz einschätzen können wieviel Herzblut Du in die Sache steckst.

In meinem Fall war mir nur nicht klar, dass es "hidden functions" gibt und nachdem ich ein paar aktiviert habe, hat das Ganze noch mehr Spass gemacht, als es vorher schon der Fall war.
Weiterhin konnte ich nicht einschätzen ob es "zu schneien aufhört", wenn Weihnachten vorbei ist? (Ob Du diese Funktion Datum gesteuert realisiert hast, war mir leider nicht klar).

vG, ein GROSSES Danke & ein schönes Weihnachtsfestchen

ProggroP, 2023-12-24:
Thank you for your great an have a great X-mas.

Dtrieb, 2023-12-24:
Stephan vielen Dank für die alternative Firmware!
Man kann echt froh sein, das du diese auf die Beine gestellt hast, denn mit ihr hat man echt einen Mehrwert!

Und zu Silvester/Neujahr gibt's dann ein Feuerwerk?(Farbige Pixel 🫣😜) 

{"firework":true} or {"firework":false}

SomeFixItDude, 2023-12-24:
@Blueforcer love the snow!  Appreciate you're work on this project.  This is the clock for my living room and notification display for my home.  Keep up the great work!  Merry Christmas

hytrek, 2023-12-24:
Hi Stephan,
das wollte ich nicht anstossen, dass sich hier in unangebrachter Weise Leute beschweren. Ich schätze es sehr, dass Du Deppen wir mir möglich machst (mit Kleinohr), so nettes Spielzeugs zum Laufen zu bringen, aber eben ohne, dass ich Programmierkenntnisse brauche. Ich habe mir die dev.json von @distanzcheck geladen und damit kann ich jetzt das machen, was ich wollte. An Euch beide herzlichen Dank! Und weiter so ... Again what learned ;-)

hytrek, 2023-12-24:
> [dev.json](https://github.com/Blueforcer/awtrix-light/files/13760908/dev.json)
> 
> Nimm mal diese dev.json und starte danach die neu.
> 
> Take this dev.json and then restart it.

Danke Dir... Nun läufts auch mit dem ein- und ausschalten bei mir!

theunknownstarman, 2023-12-25:
Weis jemand zufällig ob es möglich ist die Batterieanzeige irgendwie abzuschalten?
Ich habe die AWTRIXies am Netzteil laufen und benötige diese nicht.

Dtrieb, 2023-12-25:
> Weis jemand zufällig ob es möglich ist die Batterieanzeige irgendwie abzuschalten? Ich habe die AWTRIXies am Netzteil laufen und benötige diese nicht.

Zum Beispiel, die App runterladen und da ein bzw. ausschalten...

Blueforcer, 2023-12-25:
Oder im onscreen Menü oder per settings API

theunknownstarman, 2023-12-25:
Danke. App gekauft 🤟🏻