Blueforcer/awtrix3 #442: [FEATURE REQUEST] Multiple Time app

DizHell, 2023-12-23:
Hi,

 Would it be possible to include an option to have multiple TIME applications.
 The idea would be to go from a transition:
 TIME > TEMP > HUM > BAT > TIME > TEMP.....

 has a transition with TIME displayed between each app:
 TIME > TEMP > TIME > HUM > TIME > BAT > TIME > TEMP....

 The idea being to display the time more often.

 Another possibility in the same spirit, being able to define a much longer appearance time for TIME than other Apps

Thanks for all

///// I'm french user :

Bonjour,

Serait il possible d'inclure une option pour avoir plusieurs applications TIME.
L'idée serait de passer d'une transition :
TIME > TEMP > HUM > BAT > TIME > TEMP.....

a une transition avec TIME affiché entre chaque app :
TIME > TEMP > TIME > HUM > TIME > BAT > TIME > TEMP....

L'idée étant d'afficher plus souvent l'heure.

Autre possibilité dans le même esprit, pouvoir définir un temps d'apparition de TIME beaucoup plus long que les autres App

Merci pour tout ce que vous avez déjà accompli

Blueforcer, 2023-12-23:
Thats not possible. A app can only exists once like on a smartphone.

you have two options:
- Create your own loop automation by calling the desired app after x seconds with "switchTo" API
- You could send a json array off your "new" loop with the appnames in it to /api/apps. like ["time","hum","time","hum"]. Use this with caution since this is not an offical API

DizHell, 2023-12-26:
@Blueforcer ok thanks you for answer 👍

closed 2023-12-26 by DizHell