Blueforcer/awtrix3 #449: Display shows only upper lines after notify tests, bug or hardware failure? [BUG]

roitefr, 2023-12-26:
## Describe the bug
I have the clock for four weeks now, flashed it when I got it - everything ok then. However not time until now to test it.
Today I sent a couple of notify messages including icons and suddenly half of the pixels stayed unchanged.
Tried reboot (no cure) and re-flashed it three times, however half of the display is not working properly. Live preview still shows the correct display, see picture.
I don't know if it's a bug or a hardware problem - hardware seems weird to me since all the pixels were working - it's just that the lower rows cannot be changed any more.
[image]

a) Anybody else who has this problem?
b) Any chance to do a "factory reset"? (The reset contact on the device does nothing at all.)
 
@Blueforcer Tolle Arbeit! Auch wenn ich Ärger mit dem Gerät habe bin ich von Deinem Awtrix begeistert.
 
## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.93

Blueforcer, 2023-12-26:
Das sehe ich zum ersten Mal und auch noch keine Vergleichbaren issues erhalten. Gehe daher mal stark von einem Hardwarefehler aus.
Wie war's denn bei der originalem firmware?

roitefr, 2023-12-26:
Ja - wenn es die einzige Uhr mit dem Verhalten ist, dann ist Hardwarefehler ziemlich wahrscheinlich. Original und die letzten vier Wochen ohne MQTT-Ausprobieren war aber alles i.O.. 
Komme ich auf die Original-Firmware irgendwie zurück oder ist die ohne Copy im EEPROM gewesen und damit nun überschrieben? (Und zweite Frage: Ist mit Awtrix auch der Reset-Kontakt außer Funktion, der geht nämlich nicht?)

Blueforcer, 2023-12-26:
Hab hier ein Dump von meiner ersten:
[ulanzi_original_firmware (2).zip](https://github.com/Blueforcer/awtrix-light/files/13772935/ulanzi_original_firmware.2.zip)
[image]

Der Resetbutton hat glaub ich auch bei der originalen firmware keine funktion gehabt. Weiß gar nciht ob der überhaupt verbunden ist. zumindest hatte ich damals keine Pin verbindung gefunden

roitefr, 2023-12-26:
Danke Dir! Werde es morgen mal probieren und habe parallel bei Ulanzi einen service request aufgemacht. Jetzt aber erst einmal weiter Weihnachten feiern, wir beide! ;) Frohes Fest!

roitefr, 2023-12-27:
Closed - it IS a hardware bug and _not_ related to Awtrix. Ulanzi is going to replace the clock, they responded immediately after my support request, good company.

@Blueforcer Danke Dir für Deine Unterstützung, kommt gut ins nächste Jahr.

closed 2023-12-27 by roitefr