Blueforcer/awtrix3 #451: Units description for temp_offset

electric-m, 2023-12-27:
Hey there,

on the hidden features page, the the temp_offset variable is found.
Although I can implement it, the unit of the float values are unclear to me.

Default says -9? I don't get it.

I thought about it like this: 25 is the actual temp. If I define an offset of -10, the new temp should be 24, right?

Blueforcer, 2023-12-27:
The offest is -9 by default. If you change it it will be overwritten.
 
So if awrix shows 25 now, the temp sensor actually measure 34.

Should be more clear if you first set it to zero, and then calculate your desired offset

electric-m, 2023-12-27:
OK, thanks, makes totally sense. I just couldn't believe, how far off the sensor was from the actual temp (which I verified through multiple other sensors...).

Thanks anyway!

Blueforcer, 2023-12-27:
You will never get an accurate temperature out of this thing. Because the sensor sits in the case. To the matrix, charging circuit will affect the measurement

electric-m, 2023-12-27:
Yeah, I figured by now. Nonetheless, as I didn't by it for the temperature sensor, I am still pleased, even more by the great Awtrix firmware.. !