Blueforcer/awtrix3 #452: Hard coded sound file extension

Brunas, 2023-12-29:
Hi,

I think it's not really correct to have such statement in Sounds section in documentation:
[image]

and have `.txt` hard coded in the [code](https://github.com/Blueforcer/awtrix-light/blob/326184127509ff86074ef78ea7014debf541cb91/src/PeripheryManager.cpp#L318)

I've spent too much time on trying to find why api or mqtt sound request is not working when rtttl works.

Blueforcer, 2023-12-29:
Just ask if you have problems.
There is no extension required to keep the JSON small as possible.

Blueforcer, 2023-12-29:
> Just ask if you have problems.
> There is no extension required to keep the JSON small as possible.

Same also applies to icons.

closed 2023-12-29 by Blueforcer