Blueforcer/awtrix3 #454: Make overlay animation "soft" configurable

luebbe, 2023-12-29:
# Feature Request

I really like the snowfall overlay animation that has been added in 0.92 and I'm a bit sad to see that it is opt in now in 0.93 ;-)
Instead of making this a "hard" configuration via .json, how about making it a "soft" configuration via MQTT/HTTP api?

This would allow the user to turn the animation on or off and more animations could be added over time. Like:
- off
- snow (falls vertically, as it is now)
- snowstorm (falls almost diagonally and maybe more and faster)
- rain (like snow but faster blue drops and maybe more)
- rainstorm (like rain but almost diagonally)
- thunderstorm (like rain/rainstorm but with occasional flashes)

just venting ideas here. The user could connect to his preferred weather service and set the animation accordingly.

Happy new year everybody

MGKQB, 2023-12-30:
Imho an option in the smartphone app would be a nice solution. Doesn't require resources on the AWTRIX, easy to use and encourages people to buy it 😉

golles, 2023-12-30:
I think it was a good decision to make it opt-in instead of opt-out, but I wonder if the snow effect couldn't be added as a real [effect](https://blueforcer.github.io/awtrix-light/#/effects)? In that case, it can be used as a [background in apps or notifications](https://blueforcer.github.io/awtrix-light/#/api?id=custom-apps-and-notifications) through the API

Blueforcer, 2023-12-31:
> Imho an option in the smartphone app would be a nice solution. Doesn't require resources on the AWTRIX, easy to use and encourages people to buy it 😉

It would require the same resources, because the App only communicates with the official API wich needs to be developed anyway.

Blueforcer, 2023-12-31:
@luebbe I also had a similar idea, but the question is if this overlay still needs to be in the foreground.
When a thunderstorm is running, I imagine that you can barely see the actual content of the apps.

luebbe, 2023-12-31:
I'd imagine the thunderstorm being mostly rain with the occasional random flash every 10-15 seconds. 
But maybe raindrops are already too distracting. The snowfall now ist just perfect 👌

commit d9ad8e8d (2024-03-12): v0.95

- Rebranding AWTRIX Light to AWTRIX 3

- **Enhancements:**
  - Implemented median and mean filters for more accurate processing of raw data from LDR and BATT sensors.
  - Added Cyrillic support, limited to UPPERCASE letters. Issue #467 closed. 🇷🇺
  - Introduced overlays for weather animations accessible via the /settings API or customapp/notifications. Available overlays: clear, snow, drizzle, rain, storm, thunder, frost.  closes #454

- **Features:**
  - Added AP Mode timeout configuration via dev.json. ⏱️
  - The hostname isnt the same as the MQTT prefix anymore.  Instead you can set  [... cut]
commit 636220ff (2024-03-12): V0.95

- Rebranding AWTRIX Light to AWTRIX 3

- **Enhancements:**
  - Implemented median and mean filters for more accurate processing of raw data from LDR and BATT sensors.
  - Added Cyrillic support, limited to UPPERCASE letters. Issue #467 closed. 🇷🇺
  - Introduced overlays for weather animations accessible via the /settings API or customapp/notifications. Available overlays: clear, snow, drizzle, rain, storm, thunder, frost.  closes #454

- **Features:**
  - Added AP Mode timeout configuration via dev.json. ⏱️
  - The hostname isnt the same as the MQTT prefix anymore.  Instead you can set  [... cut]
commit f0d82d13 (2024-03-12): v0.95

- Rebranding AWTRIX Light to AWTRIX 3

- **Enhancements:**
  - Implemented median and mean filters for more accurate processing of raw data from LDR and BATT sensors.
  - Added Cyrillic support, limited to UPPERCASE letters. Issue #467 closed. 🇷🇺
  - Introduced overlays for weather animations accessible via the /settings API or customapp/notifications. Available overlays: clear, snow, drizzle, rain, storm, thunder, frost.  closes #454

- **Features:**
  - Added AP Mode timeout configuration via dev.json. ⏱️
  - The hostname isnt the same as the MQTT prefix anymore.  Instead you can set  [... cut]
closed 2024-03-12 by Blueforcer