Blueforcer/awtrix3 #456: reuse ulanzi case and display

fragolinux, 2023-12-30:
hi, not really a sw request, but don't know where else to ask: I, as many others, had my ulanzi clock completely frozen and blocked after a few months usage... i saw threads about changing an ldo voltage regulator to fix it, i tried but no luck... so, i was thinking if is there a way to reuse the ulanzi case and display, adding an esp32 with some other hardware in it, to have a similar functionality to the original one...

is the hardware design fully documented, i mean, what gpio is used of what? I think yes, based on blakadder work for tasmota on this device... any suggestion to a way to not throw away a partially broken good piece of hardware? Or any suggestion to regain access to the original? the ldo regulator swap didn't work for me...

thanks, and happy new year!

Blueforcer, 2023-12-31:
You can find the pinout in the documentation 
https://blueforcer.github.io/awtrix-light/#/hardware

There are also some threads about successful conversations from other users at my discord server.

fragolinux, 2023-12-31:
@Blueforcer nice, thanks! just joined!

trip5, 2024-01-04:
@fragolinux As you see over on the other thread, I wasn't happy with how my LDO "fix" worked.  No battery and I have to press 2 buttons just to turn it on... I'm planning on swapping out the mainboard for a esp32 of some kind, while re-using the buttons and perhaps the light sensor (LDR).

I made notes months ago and haven't gotten around to actually doing it... and it looks like my notes aren't making a lot of sense.  So this thread on discord looks useful: https://discord.com/channels/546407049148366859/1090595658437767279/threads/1101071217986900049

Blueforcer, 2024-03-08:
For further hardware discussion please move to my discord

closed 2024-03-08 by Blueforcer