Blueforcer/awtrix3 #457: No Sound[BUG]

dl1ekm, 2024-01-01:
# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

dl1ekm, 2024-01-01:
Why completed?

Blueforcer, 2024-01-01:
Because at there was no single description of your problem. 
Now it's edited

Blueforcer, 2024-01-28:
What firmware youre using?

Blueforcer, 2024-03-06:
is this fixed?

dl1ekm, 2024-03-06:
I haven given up with 0.94 and desolded the speaker :-) Last idea - is an ACTIVE buzzer (just needs 3,3V) or a passive one (controlled with PWM) required? My current buzzer is only "clicking" - in the past  (older FW) I tested both variants without success. Should I give it again a try?

Blueforcer, 2024-03-06:
Ulanzi firmware only supports a passive piezo buzzer.

Awtrix2_upgrade only supports dfplayer. But this firmware is made for users of the old awtrix so they don't need to rebuild

closed 2024-03-17 by Blueforcer