Blueforcer/awtrix3 #458: [BUG] No sound with ESP Wroom 32 Board

dl1ekm, 2024-01-01:
# Bug report

I think I saved my report in 457 in an undefined state. Sorry for that. If I made something else wrong please let me know.

I installed the default Ulanzi binary V0.91 on this board as descripted in the documentation and used the hardware-description to wire my own Awtrix panel. All is working using the default Ulanzi/ESP ports - my buttons, light, sensors... but not the buzzer. It should be connected to GPIO15 - but even with my scope I see nothing here.

## Additional information

- Devices involved:
  - Model: ESP 32 Board with Matrix-Panel from Ali Express (mode 2)
  - AWTRIX-LIGHT version: [ v.0.91 ]

## To Reproduce

I put a custom app in the custom app folder displaying some text. Additionally I set the parameter "sound" : "starwars" in the json. I uploaded a sample melodie (from internet and checked with the given links), saved as starwars.txt.

## Expected behavior
Shouldn't I hear the sound while text is scrolling?

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2024-01-01:
Sound isn't supported by customs pls, only notifications. See
https://blueforcer.github.io/awtrix-light/#/api?id=json-properties

dl1ekm, 2024-01-01:
Ok - seen.... but what about the sound option "bootsound" in the dev,json? Shouldn't that work?

Blueforcer, 2024-01-01:
That's currently a bug and will be fixed with the next version

dl1ekm, 2024-01-01:
Ok - super. Thanks (also for this great project!).

closed 2024-03-06 by Blueforcer