Blueforcer/awtrix3 #461: [BUG] Automatically in Setup Menu on startup

dl1ekm, 2024-01-02:
# Bug report

## Describe the bug

When powering up Awtrix on a standard ESP it goes into menu first - showing "BRIGHT". I have to press a button to continue.
Shouldn't it show the apps automatically?

## Additional information

- Devices involved: 
  - Model: ESP Wroom 32
  - AWTRIX-LIGHT version: [ v0.91 ]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2024-01-02:
That's not normal, seems like a wiring issue and your middle button gives a signal by default

dl1ekm, 2024-01-02:
Thanks for feedback. I will investigate.

dl1ekm, 2024-01-04:
Can you please check? Miercke button seems  to be ok and is working further on. Maybe a missing internal pullup and the original Ulanzi has HW resistors? Or are HW pullups required?

dl1ekm, 2024-01-20:
I tried again with another touch sensor. But it always greets me with the BRIGHT menu.
I'm using an ESP32 WROOM module.
If I disconnect the middle-touch-button there is no menu entry.
Maybe the SW is checking this button too early or ....?

dl1ekm, 2024-01-28:
Hi, I enabled the debug mode and got this at startup:
```

[134282] [setCurrentApp]: Publish current app Time
[2171] [setup]: Setup periphery
[2364] [loadSettings]: Webserver configuration loaded
[2365] [setMatrixLayout]: Set matrix layout to 2
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to xxxx
....[4222] [setup]: My IP: 192.168.1.149
[8529] [setup]: Webserver loaded
[17678] [select_button_pressed_long]: Select button pressed long
```

But the button IS NOT PRESSED. I changed the touch buttons - no change. There is not red light on the touch buttons. But later on the middle button is working fine. Strange is: If I pull of the connection cable from the touch button it doesn't start the menu.

Blueforcer, 2024-01-28:
do a solder bridge on your middle touch button:
[image]
,

dl1ekm, 2024-01-28:
No, as listed above  I use an ESP32 DEV board and firmware is upgraded to 0.94.
I'm very sure it's not a hardware problem: If I change the wiring of the 3 touch buttons (mixing the buttons to different GPIOs) I always get the message at startup for button_select (although other cable / pin).
Seems the "onPressFor" of EasyButtons need some special behavior? Or maybe some pullups/downs on GPIO27 (assigned to button_select in the firmware)?

[2172] [setup]: Setup periphery
[2365] [loadSettings]: Webserver configuration loaded
[2366] [setMatrixLayout]: Set matrix layout to 2
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to xxx
.[3323] [setup]: My IP: 192.168.1.149
[7631] [setup]: Webserver loaded
**[17228] [select_button_pressed_long]: Select button pressed long**

Attached the over all fine working assembly and a touch button for reference view. I think the GPIO 27 must be handled different on ESP board.
[image]

Blueforcer, 2024-01-28:
Unfortunately this is not the answer of my question.
The relase tab in github provides 2 different Firmwares, as well as the online flasher

**old_awtrix2_conversion_0.94.bin**
and
**ulanzi_TC001_0.94.bin**

Wich one youre using?

dl1ekm, 2024-01-29:
Sorry. I thought this was clear :-) I use the current ulanzi firmware and used this information:
[image]

It's working perfect with 2 eecptions: 
a) no sound - just "ticks". tested with active and passive buzzers
b) on startup "button_select long pressed" - although not touched . going into menu. Later on the menu control works fine.

For testing I will do to "upgrade version soldering" for the button - maybe it will help.

dl1ekm, 2024-01-29:
The bridge on "A" for the middlebutton seems promissing. Let me test 1 day and I'll give more feedback.

dl1ekm, 2024-01-30:
Hi, although I use the "normal" awtrix-light firmware on a plain ESP32 the "solder bridge A on middle-button" fixes the issue (documented only for the upgrade from an Wemos).
My suggestion: please add this solder hint to your documentation page "hardwaer -> diy" shown above
(https://blueforcer.github.io/awtrix-light/#/hardware)
Thanks for support!

closed 2024-03-06 by Blueforcer