Blueforcer/awtrix3 #463: [FEATURE REQUEST] 3 More Indicators Please

DaveTiff, 2024-01-03:
Love this display and its good using it with Home Assistant, but I could do with more indicators.

I use the indicators to show when the upstairs heating is On, Hot Water Heating is On & downstairs heating is On, that's all 3 used but I would also like to display when my cheap rate electricity kicks in and when my car is on charge.

I believe there is room on the display, just means changing the shape of the existing indicators and shifting everything to the right by 1 pixel.

I have drawn up what I believe is a solution.

Add any other context or screenshots about the feature request here.
[image]

DaveTiff, 2024-01-09:
Just a thought and maybe easier alternative is if you have the week days disabled, then allow each of the week days to be individually addressable, that way you would get 7 more indicators.

Blueforcer, 2024-01-09:
But only in the time app. 
To move all content 3 pixels to the right will also shorten the available content space which is already pretty short. And this will also affect other functions like the progressbar.

DaveTiff, 2024-01-09:
Happy with only in the time app, that is all that I have constantly displayed, except for when a message is sent.

Blueforcer, 2024-01-09:
But very few people do, so the effort involved is quite high and out of all proportion. It would therefore be better if you simply created a custom app according to your preferences

closed 2024-01-09 by Blueforcer

DaveTiff, 2024-01-11:
Totally understand, would love to create a custom app, but not sure where to start. But many thanks for the great work you have done.