Blueforcer/awtrix3 #464: [BUG] When using Display Text in Colored Fragments in a custom app, you can't go back

DennisDeKoster, 2024-01-03:
# Bug report

## Describe the bug

When using a custom app and send text in colored fragments, you cannot change back to a single color.
The other way around is not a problem: When displaying text in single color, you can update the app with colored fragmented text without issue.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.91

## To Reproduce
Create or update a custom app with colored text fragments:
```
{
    "text": [{
        "t": "example text 1",
        "c": "FFFFFF"
    },
    {
        "t": " example text 2",
        "c": "FF0000"
    }]
}
```

Then update the app with normal text:
```
{
    "text": "example text 3",
    "color": "FFFFFF"
}
```
Which doesn't work, the app displays the colored fragments text.

## Expected behavior

I expect to display the text as sent in the payload, regardless of the previous.

## Screenshots

-

## Logs

```text
-
```

## Additional context

-

MarcoRemy, 2024-02-15:
I can confirm this. Text can only be changed with fragments.
v0.94

Blueforcer, 2024-02-15:
It's fixed in the latest beta.

closed 2024-03-12 by Blueforcer