Blueforcer/awtrix3 #465: [BUG] let_it_snow not working

dl1ekm, 2024-01-03:
# Bug report

## Describe the bug

This option in the dev.json has no effect.

## Additional information

- Devices involved:
  - Model: ESP Wroom 32
  - AWTRIX-LIGHT version: [ v0.91]

## To Reproduce

Put the line

 "let_it_snow": true
In the dev.json file.Restart.

## Expected behavior

Snow as background effect / global overlay.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

RDG88, 2024-01-04:
Please show the content of your dev.json

dl1ekm, 2024-01-04:
After update to v0.94 it's working.I
Thanks for support!

closed 2024-01-04 by dl1ekm