Blueforcer/awtrix3 #466: [BUG] Stuck in AP mode depending on WiFi

limes007, 2024-01-04:
# Bug report

## Describe the bug

I flashed Version 0.94 to my TC001. If I connect it to my guest WiFi it will work fine showing time, date and so on. If I connect to my home WiFi it will connect (web interface is available inside network) but clock remains at "AP mode". Both WiFi are provided by the same router. I tried differend reflashes and restarts, its always the same.

Interesting observation: The second log entry "Connecting to" shows the SSID appended by the password for my home WiFi, but not for the guest WiFi (see below).

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.94
  - WiFi router: FRITZ!Box 7520

## To Reproduce

Steps to reproduce the behavior:
* Flash awtrix 0.94
* Open 192.168.4.1
* Enter home WiFi and confirm

## Expected behavior

AP mode will be disabled for my home WiFi also

## Screenshots

n/a

## Logs

Home WiFi
```text
Connecting to QXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
*.**.**.**.**.**.*
Connected to Wifi! IP address: 192.168.xxx.xxx
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to QXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXPASSWORD
...................................................
AP mode.
Server IP address: 192.168.4.1
```

Guest WiFi
```text
Connecting to NXXXXXXXXXXXXXXXXXXXXXXXXXX
*.**.**.**.**.**.**.*
Connected to Wifi! IP address: 192.168.xxx.x
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to NXXXXXXXXXXXXXXXXXXXXXXXXXX
```

## Additional context

I use very long SSID for the home WiFi (32 char) but that's inside the specification. May this be an issue here?

limes007, 2024-01-05:
It gave me no rest, so I debugged it myself and found something, but not the root cause. As far as I can see, it is a problem with storing/retrieving the WiFi settings when the SSID is 32 chars.

I have created a patched firmware that truncates the SSID in FSWebServer::startWiFi() to 32 chars (maximum according to spec) if it is longer. With this patch it works fine for me with my home WiFi too.

```udiff
diff --git a/lib/webserver/esp-fs-webserver.cpp b/lib/webserver/esp-fs-webserver.cpp
index 326f247..706d368 100644
--- a/lib/webserver/esp-fs-webserver.cpp
+++ b/lib/webserver/esp-fs-webserver.cpp
@@ -183,6 +183,13 @@ IPAddress FSWebServer::startWiFi(uint32_t timeout, const char *apSSID, const cha
     _ssid = reinterpret_cast<const char *>(conf.sta.ssid);
     _pass = reinterpret_cast<const char *>(conf.sta.password);

+    char * my_ssid;
+    my_ssid = new char[32+1];
+    strncpy(my_ssid, _ssid, 32);
+    my_ssid[32] = '\0';
+    if (strlen(my_ssid) < strlen(_ssid))
+        _ssid = my_ssid;
+
     if (strlen(_ssid) && strlen(_pass))
```

pflavio, 2024-02-19:
Any chance you can upload that branch or request a pull? I have a similar issue, although it's the passwords length (32 characters), not the SSID (which is much shorter). I tried it on four networks (all 32 character passwords but varying SSID lengths).

limes007, 2024-02-19:
Of course. I have created a pull request see #494.

But I think you have a different problem. My WiFi password is 63 chars long and I don't had any problems with length of password. You might have some ugly special characters in your password, which could cause issues. I have some, but nothing very special.

phil0412, 2024-02-20:
I tried again with a password that only has letters and numbers. 32 characters. The SSID is 10 characters including three dashes (-), which shouldn't be a problem I think?

limes007, 2024-02-20:
I think this should be not the problem.

wolfdave, 2024-03-04:
Same here. If I use same Wifi-Name and WPA-Key on my phone by opening a hotspot it works. When using a raspberry hotspot with the same data it doesn't. It remains in AP-Mode.
Any Ideas?

wolfdave, 2024-03-04:
It's also 2,4 Ghz like the one the phone. 
Sorry, I forgot to mention. Also the encryption method is the same. I have tried another TC001 with no luck. What I have noticed: on the first connection after flashing the device the images for the app stores in the bottom of the first page are present. After the first connect to a wifi and switching again in the ap mode they disappear. I don't know if this is important or not.

pflavio, 2024-03-05:
> on the first connection after flashing the device the images for the app stores in the bottom of the first page are present. After the first connect to a wifi and switching again in the ap mode they disappear. I don't know if this is important or not.

They're probably being loaded from the web and when there is no internet connection the second time you try to connect to the device that reloading fails

wolfdave, 2024-03-06:
I don't think so, because the first time after flashing the device they are present. At this time there is no internet connection until you enter the credentials of your wifi. But I don't think it is important. I just wanted to mention it. 
I'm going to check the connection using the Debug console in the flasher tool today.

Blueforcer, 2024-03-17:
> Of course. I have created a pull request see #494.
>
> But I think you have a different problem. My WiFi password is 63 chars long and I don't had any problems with length of password. You might have [...]

So your issue is solved with latest firmware, right? Please close the issue if so

limes007, 2024-03-18:
Yes, closed

closed 2024-03-18 by limes007