Blueforcer/awtrix3 #467: [FEATURE REQUEST] Cyrillic font (ready-made font and code)

geo-mesh, 2024-01-04:
At the moment, AWTRIX does not support the Cyrillic font. I would like to see the Cyrillic alphabet in future firmware releases, for this I created a "UPPERCASE font" of 5px and a "lowercase font" of 4px. I also wrote the HEX-code for these fonts (only the addresses need to be edited). Please let me know if I can now count on Cyrillic support in future projects?

 [image]
 # AWTRIX Cyrillic UPPERCASE font
 
    0xE0, 0xA0, 0xE0, 0xA0, 0xA0,             /*[33]   0x41 А */ 
    0xE0, 0x80, 0xE0, 0xA0, 0xC0,             /*[34]   0x42 Б */ 
    0xC0, 0xA0, 0xE0, 0xA0, 0xC0,             /*[35]   0x43 В */ 
    0xE0, 0x80, 0x80, 0x80, 0x80,             /*[36]   0x44 Г */ 
    0x60, 0x50, 0x50, 0x50, 0xE8,             /*[37]   0x45 Д */
    0xE0, 0x80, 0xC0, 0x80, 0xE0,             /*[38]   0x46 Е */
    0xA0, 0x00, 0xE0, 0x80, 0xC0, 0x80, 0xE0, /*[39]   0x47 Ё */
    0xA8, 0xA8, 0x70, 0xA8, 0xA8,             /*[40]   0x48 Ж */
    0xC0, 0x20, 0x60, 0x20, 0xE0,             /*[41]   0x49 З */
    0xA0, 0xA0, 0xA0, 0xA0, 0xC0,             /*[42]   0x4A И */
    0x40, 0x00, 0xA0, 0xA0, 0xA0, 0xA0, 0xC0, /*[43]   0x4B Й */
    0xA0, 0xA0, 0xC0, 0xA0, 0xA0,             /*[44]   0x4C К */
    0x60, 0xA0, 0xA0, 0xA0, 0xA0,             /*[45]   0x4D Л */
    0x88, 0xD8, 0xA8, 0x88, 0x88,             /*[46]   0x4E М */
    0xA0, 0xA0, 0xE0, 0xA0, 0xA0,             /*[47]   0x4F Н */
    0xE0, 0xA0, 0xA0, 0xA0, 0xE0,             /*[48]   0x50 О */
    0xE0, 0xA0, 0xA0, 0xA0, 0xA0,             /*[49]   0x51 П */
    0xE0, 0xA0, 0xE0, 0x80, 0x80,             /*[50]   0x52 Р */
    0xE0, 0x80, 0x80, 0x80, 0xE0,             /*[51]   0x53 С */
    0xE0, 0x40, 0x40, 0x40, 0x40,             /*[52]   0x54 Т */
    0xA0, 0xA0, 0xE0, 0x40, 0x40,             /*[53]   0x55 У */
    0xF8, 0xA8, 0xF8, 0x20, 0x20,             /*[54]   0x56 Ф */
    0xA0, 0xA0, 0x40, 0xA0, 0xA0,             /*[55]   0x57 Х */
    0xA0, 0xA0, 0xA0, 0xA0, 0xD0,             /*[56]   0x58 Ц */
    0xA0, 0xA0, 0x60, 0x20, 0x20,             /*[57]   0x59 Ч */
    0xA8, 0xA8, 0xA8, 0xA8, 0xF8,             /*[58]   0x5A Ш */
    0xA8, 0xA8, 0xA8, 0xA8, 0xF4,             /*[58]   0x5A Щ */
    0xC0, 0x40, 0x70, 0x50, 0x70,             /*[58]   0x5A Ъ */
    0x88, 0x88, 0xE8, 0xA8, 0xE8,             /*[58]   0x5A Ы */
    0x80, 0x80, 0xE0, 0xA0, 0xE0,             /*[58]   0x5A Ь */
    0xC0, 0x20, 0x60, 0x20, 0xC0,             /*[58]   0x5A Э */
    0xB8, 0xA8, 0xE8, 0xA8, 0xB8,             /*[58]   0x5A Ю */
    0xE0, 0xA0, 0x60, 0xA0, 0xA0,             /*[58]   0x5A Я */

[image]
 # AWTRIX Cyrillic lowercase font

    0xC0, 0x60, 0xA0, 0xE0,             /*[33]   0x41 а */ 
    0x60, 0xC0, 0xA0, 0xE0,             /*[34]   0x42 б */ 
    0x40, 0xA0, 0xC0, 0xA0, 0xC0,       /*[35]   0x43 в */ 
    0xE0, 0xA0, 0x80, 0x80,             /*[36]   0x44 г */ 
    0x60, 0x50, 0x50, 0xE8,             /*[37]   0x45 д */
    0x60, 0xA0, 0xC0, 0x60,             /*[38]   0x46 е */
    0xA0, 0x00, 0x60, 0xA0, 0xC0, 0x60, /*[39]   0x47 ё */
    0xA8, 0x70, 0xA8, 0xA8,             /*[40]   0x48 ж */
    0x20, 0x40, 0x20, 0xE0,             /*[41]   0x49 з */
    0xA0, 0xA0, 0xA0, 0xC0,             /*[42]   0x4A и */
    0x40, 0x00, 0xA0, 0xA0, 0xA0, 0xC0, /*[43]   0x4B й */
    0xA0, 0xC0, 0xA0, 0xA0,             /*[44]   0x4C к */
    0x60, 0xA0, 0xA0, 0xA0,             /*[45]   0x4D л */
    0x88, 0xD8, 0xA8, 0x88,             /*[46]   0x4E м */
    0xA0, 0xA0, 0xE0, 0xA0,             /*[47]   0x4F н */
    0x40, 0xA0, 0xA0, 0x40,             /*[48]   0x50 о */
    0xE0, 0xA0, 0xA0, 0xA0,             /*[49]   0x51 п */
    0xC0, 0xA0, 0xE0, 0x80,             /*[50]   0x52 р */
    0x60, 0x80, 0x80, 0x60,             /*[51]   0x53 с */
    0xE0, 0x40, 0x40, 0x40,             /*[52]   0x54 т */
    0xA0, 0xE0, 0x20, 0xC0,             /*[53]   0x55 у */
    0x78, 0xA8, 0xF0, 0x20,             /*[54]   0x56 ф */
    0xA0, 0x40, 0xA0, 0xA0,             /*[55]   0x57 х */
    0xA0, 0xA0, 0xA0, 0xD0,             /*[56]   0x58 ц */
    0xA0, 0xA0, 0x60, 0x20,             /*[57]   0x59 ч */
    0x88, 0xA8, 0xA8, 0x70,             /*[58]   0x5A ш */
    0x88, 0xA8, 0xA8, 0x74,             /*[58]   0x5A щ */
    0xC0, 0x60, 0x50, 0x60,             /*[58]   0x5A ъ */
    0x88, 0xC8, 0xA8, 0xC8,             /*[58]   0x5A ы */
    0x80, 0xC0, 0xA0, 0xC0,             /*[58]   0x5A ь */
    0xC0, 0x20, 0x60, 0x20, 0xC0,       /*[58]   0x5A э */
    0x98, 0xA8, 0xE8, 0xB0,             /*[58]   0x5A ю */
    0x60, 0xA0, 0x60, 0xA0,             /*[58]   0x5A я */

10der, 2024-01-06:
@geo-mesh maybe I something miss here. 
You provided ibm-866 encoding instead of UTF-16 encoding, what using right now in ./Awtrix Font.h

for example, "я" = 0x044F but not 0xEF

geo-mesh, 2024-01-06:
> @geo-mesh maybe I something miss here. You provided ibm-866 encoding instead of UTF-16 encoding, what using right now in ./Awtrix Font.h
> 
> for example, "я" = 0x044F but not 0xEF

Thanks for the information. I'll redo the code to UTF-16.

geo-mesh, 2024-01-06:
> @geo-mesh maybe I something miss here. You provided ibm-866 encoding instead of UTF-16 encoding, what using right now in ./Awtrix Font.h
> 
> for example, "я" = 0x044F but not 0xEF

I have already fixed this and added a new code with UTF-16. Thanks again.

10der, 2024-01-06:
@geo-mesh Sir, can I ask you a little deal. Can you reserve 3 x 2 chars as well, please?
I'll patch it after merge.
so:
```
0x00, 0x00, 0x00, 0x00, 0x00, 	  /*[000]   0x0404 Є */
0x00, 0x00, 0x00, 0x00, 0x00,        /*[000]   0x0406  І */
0x00, 0x00, 0x00, 0x00, 0x00,        /*[000]   0x0407  Ї */
```

```
0x00, 0x00, 0x00, 0x00, 0x00, 	/*[000]   0x0454 є */
0x00, 0x00, 0x00, 0x00, 0x00,      /*[000]   0x0456  і */
0x00, 0x00, 0x00, 0x00, 0x00,      /*[000]   0x0457  ї */
```

Thank you!

geo-mesh, 2024-01-06:
> @geo-mesh Sir, can I ask you a little deal. Can you reserve 3 x 2 chars as well, please? I'll patch it after merge. so:
> 
> Thank you!

No problems. I can also write the code for these letters. The only thing I'm not sure yet is whether it's worth leaving serifs on the font? Serif looks beautiful, sans serif looks compact.

[image]

[image]

10der, 2024-01-06:
wow! das ist fantastisch, thank you so much!

10der, 2024-01-06:
btw. compact - the best, IMHO

megadimich, 2024-01-06:
This is a great idea! I'm waiting for a full-fledged implementation in awtrix!

10der, 2024-01-19:
@geo-mesh 
glyphs after "в" wrong!
```
    {1074, 8, 5, 4, 0, -5}, /*[238]   0x432 в */
    {1081, 8, 4, 4, 0, -4}, /*[239]   0x433 г */
```
1074 + 5 = 1079 not 1081
all chars after - affected and looks like rubbish on the screen

10der, 2024-01-19:
The python app emulate loading font and write font chars on canvas.
as u can see BIG cyr char looks OK
small - wrong

10der, 2024-01-19:
[image]

10der, 2024-01-19:
[sources](https://gist.github.com/10der/6eaf4fd6a365121c877dfccbc13257cc)

megadimich, 2024-01-19:
Great job! When can we see this on awtrix watch?

10der, 2024-01-19:
when topic starter help me to fix font. @geo-mesh

10der, 2024-01-20:
@geo-mesh https://discord.com/channels/546407049148366859/833683595670454443
done (as draft)

geo-mesh, 2024-01-21:
[10der](https://github.com/10der) Thank you for your interest and work. At some point, I realized that it was not enough just to add the HEX code of the font to the file "AwtrixFont.h" so that it would work and I stopped being interested in it.
I do see a mistake with the letter "з". The correct code for it is: 
`0xC0, 0x20, 0x40, 0x20, 0xE0, /*[241] 0x437 з */`
perhaps because of this, the glyphs may shift. I have already fixed the code in the header of the theme.
What software do you use to test the font on a PC? Is this Visual Studio Code + Python + add-on? I would like to test this on my PC before doing the firmware. I've been working on the font a lot, but I don't know how to test it on a PC and because this, interest is lost.

10der, 2024-01-24:
@geo-mesh https://tchapi.github.io/Adafruit-GFX-Font-Customiser/
btw, you can close this FR.

closed 2024-03-12 by Blueforcer