Blueforcer/awtrix3 #470: [FEATURE REQUEST] Add Restart Timer Option to Prevent AP Mode

pablogventura, 2024-01-08:
# Feature Request

### Issue

Occasionally, there are power outages in my house. When power is restored, Awtrix enters AP mode before my router can provide Wi-Fi.

### Proposed Feature

I suggest adding an option in `dev.json` for a timeout in minutes to restart in case the clock enters AP mode. Setting it to -1 could be the default, maintaining the current implementation without any timeout for AP mode.
### New feature

I think would be useful to have a option in dev.json that is a timeout in minutes to restart in case the clock enter in the AP mode. Maybe -1 could be the default, leaving AP mode without any timeout like it's today implemented.

Blueforcer, 2024-01-08:
There is already a timeout of 15s:
https://github.com/Blueforcer/awtrix-light/blob/615ee24e94f8d0ff1ff0f3bc5789a4b535c42fa1/src/ServerManager.cpp#L196

i will try to make it adjustable

pablogventura, 2024-01-08:
Do you mean the timeout to enter AP mode? I meant a timeout to reset the esp after a certain timeout when no client is connected to the AP mode.

By the way, thanks a lot for your work!

robercy, 2024-01-10:
> Do you mean the timeout to enter AP mode? I meant a timeout to reset the esp after a certain timeout when no client is connected to the AP mode.
> 
> By the way, thanks a lot for your work!

Hi. This is my problem too.  I think this is a great solution to avoid a manual restart. Or maybe, only enter in AP mode if press one of three buttons.

Blueforcer, 2024-02-13:
Current beta allows to set a custom timeout before starting the AP.

topcaser, 2024-02-19:
Hi, I am also using your Firmware on my Ulanzi TC001 - the version is 0.94. I completely switching the watch off at night by using the left and right arrow key. Furthermore I disconnect it from the power supply. If I switch it on in the morning, the watch is going to display AP mode. This is totally reproducable although my WIFI is stable 24/7. I first have to connect to the AP, open the browser and store the WIFI cridentials again. Is my unit faulty or is this a firmware bug?

Blueforcer, 2024-02-19:
>  Is my unit faulty or is this a firmware bug?

If it were a firmware bug, I would have several such issues from the now over 4000 users. But so far I haven't heard anything like that. Try switching on awtrix next to your router. Maybe the reception of the small ESP32 is simply too weak and the connection takes longer than the 15s timeout.
if it still doesn't work, then I would start by completely erasing and reflashing the ESP, otherwise I would assume a hardware defect.

topcaser, 2024-02-19:
OK, thanks for that answer. I can't believe that the WiFi signal is too weak since I built up my new home with Ubiqiti Unify 6 pro on each floor. But I will try that out to switch it on directly under the AP. I will update here as soon as I found something. Thanks anyway for your answer.

Blueforcer, 2024-02-19:
Ah, Ubiqiti.. ok this is a known thing with ESP32 in general:

From my discord channel:

**After chatting with RobG, it appears that we both saw this on Ubiquiti routers. Whenever the router "optimizes" the channel that your 2.4ghz wifi is on, and selects channel 11, the Ulanzi experience frequent wifi dropouts. Changing to another channel seems to resolve the issue.**

topcaser, 2024-02-19:
Haha, you are right, at least I was able to switch it back on on my first floor where channel 5 dominates. On the ground floor channel 11 is present. OK, if this behavior is stable, it's a workaround with which I can live with. Thanks for pointing that out

topcaser, 2024-02-25:
Hi again, it turned out that channel 5 on first floor isn't reliable available in the morning. This is something which I have to debug on my site. But there is something which I don't understand. If I switch on the beast it says AP mode. I tried to connect to the URL (in my case 192.168.178.95, without logging into the AP WiFi) and voila I can set the WiFi credentials again. Does this mean that the device only forgets the WiFi credentials? Why is there not the necessity to log in the provided AP?

Blueforcer, 2024-06-11:
The AP mode is always started when no connection to the Wifi can be established. Regardless of whether incorrect data, little reception wtc