Blueforcer/awtrix3 #474: [FEATURE REQUEST] Make settings available as configuration items in HA

hmmbob, 2024-01-10:
# Feature Request

### Is your feature request related to a problem?
I don't like that I can't set the brightness level directly from Home Assistant (I can only switch between manual and automatic mode). There are more settings in https://blueforcer.github.io/awtrix-light/#/api?id=change-settings that would be great to directly set from HA.

### Describe the solution / feature you'd like
It would be great if all (or most) settings available through https://blueforcer.github.io/awtrix-light/#/api?id=change-settings are directly available in HA (as boolean/number/whatever) so one can easily set it through HA instead of drafting a special JSON. In example if the brightness setting: a number entity with values from 0-255 (or 0%-100%) that when changed get's sent to the clock (`BRI` value). Other candidates could be `ATIME`, `TSPEED`, `WD` (boolean), `SSPEED` `TIM/DAT/HUM/TEMP/BAT`

### Describe alternatives you've considered
None

### Additional context
Probably these items should be marked as `EntityCategory.CONFIG`
https://developers.home-assistant.io/docs/core/entity#registry-properties

Blueforcer, 2024-01-10:
The problem here is that there are simply too many settings and there are more and more of them. Each entity consumes RAM on the ESP because it has to provide the JSON for the Homeassistant discovery function.
However, since Awtrix is universal and also compatible with other systems through the API, the result would be that other users would also want complete integration into their smart home system. There would be no end to the feature requests.
So unfortunately I have to decline this request. Especially as it is relatively easy to create your own entities with the desired functionality in Homeassistant by calling the API.
Maybe there will be a HACS Integration in the future.

closed 2024-01-10 by Blueforcer

hmmbob, 2024-01-10:
Understood, thanks.

but, HACS integration? Is there an integration and/or is someone working on that? Happy to help testing.

Blueforcer, 2024-01-10:
Not yet.