Blueforcer/awtrix3 #475: [BUG] Issue with default Time app and low brightness level

leinardi, 2024-01-10:
# Bug report

## Describe the bug

I just noticed an issue on the default clock when the brightness of the device is set to a very low value (e.g. on a dark room):
the selected day of the week is indistinguishable from the others (see the picture below). The issue happens only with very low levels of brightness (e.g. 1).

Would it be possible to use different colors for the days of the week and the current day of the week? this should solve the problem.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.94

## To Reproduce

Steps to reproduce the behavior:

- Set the brightness level to the minimum.

## Expected behavior

No matter what is the brightness level used, the current day of the week should be clearly distinct from the others, also for accessibility reasons. Maybe the current day could be the same red color as the top of the calendar?

## Screenshots

[image]

This picture was taken with the brightness to the minimum. With an higher value of brightness the issue doesn't happen.

For accessibility reasons would still be good to use high contrast colors between the current day of the week and the others.

Blueforcer, 2024-01-10:
That's normal because the LEDs doesn't get enough power at low brightness. So blue for example disappearing first because it needs more power than the other two.

Blueforcer, 2024-01-10:
Its already possible to change the colors?
https://blueforcer.github.io/awtrix-light/#/api?id=json-properties-1

leinardi, 2024-01-10:
Oh neat, I didn't realize, tomorrow I'll try to change it and report back 👍

leinardi, 2024-01-11:
It works like a charm: I used the `WDCA` property to set a different color!

closed 2024-01-11 by leinardi