Blueforcer/awtrix3 #480: [BUG] Wake up after sleep doesn't work reliably

limes007, 2024-01-22:
# Bug report

## Describe the bug

Wake up after sleep doesn't work reliably. Sometimes the clock wakes up after the timeout, sometimes not. I've tried different sleep times, thinking it would only work for shorter times, but it doesn't work for shorter times like 3 hours.

I'd like to try letting the clock sleep overnight. As sleeping for 12 hours did not work, I tried letting it sleep for 6 hours twice, then 4 hours three times and so on. The clock wakes up once or twice, but then it doesn't wake up again.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.94

## To Reproduce

Steps to reproduce the behavior:
- send the clock to sleep via API for some time
- clock wakes up or not

## Expected behavior

Clock wakes up every time.

## Screenshots

n/a

## Logs

How would it be possible to log this behaviour?

## Additional context

The clock is not connected to the power supply during sleep.

jlegen, 2024-12-30:
i had exactly the same problem - which for me has been 100% resolved since i do not run the Ulanzi on battery any more! 
Since i leave it powered b USB, i send it to deep sleep 2h after sunset, and it wakes up at sunrise. 
It also is set to deep sleep for 30 mins when nobody is at home, wake-up then first checks if that changed, and if not, it sleeps for another 30 mins. This now works since 3 months without any glitch...
Power consumptiom in deep sleep is 0.7W - ok for me...

Clubeddie, 2025-02-20:
@jlegen how do get hem into a deep sleep. I have mine connected at a usb adapter. Within Home Assistant i send around 23.00 an matrix off command for the night. I want to 'wake-up' him in the morning with a command matrix on. But somewhere during the night the connection is lost. Battery is no issue. But maybe a deep sleep command is different then a matrix off command from home assistant?

mabiste, 2025-03-04:
I have a similar problem. Regardless of whether I operate my Ulanzi TC001 (Awtrix 3 v0.98) with or without a connected USB cable, it switches off after a while.
I don't know if this is the 'deep sleep' mode, how is it defined or how can it be recognized?

Blueforcer, 2025-03-04:
Deep sleep is a function wich you need to activate via API. It has nothing to do with your problem.
Your power supply is simply too weak. Use a stronger one.

> I don't know if this is the 'deep sleep' mode, how is it defined or how can it be recognized?

Clubeddie, 2025-03-04:
I fixed it through an automation and send a sleep for approx 8 hours. Works perfect.

`alias: Ulanzi Klok nachtslaap
description: ""
triggers:
  - at: "23:30:00"
    trigger: time
    enabled: true
actions:
  - data:
      topic: awtrix_YOURID/sleep
      payload: "{\"sleep\": 28800}"
    action: mqtt.publish
mode: single`

Blueforcer, 2025-03-04:
That's not your actual problem. You just lower the power consumption. use a stronger power supply and you can do whatever you want

Blueforcer, 2025-03-04:
Oh sorry.. just saw the topic 🤣

limes007, 2025-03-04:
@Clubeddie Thank you. I use similar automation. But wake up don't work reliably for me. At some days it is working, at some days not. Maybe its the battery.

Clubeddie, 2025-03-04:
@limes007 i find the battery very unreliable. If the screen is off and there is no power it's pretty quick empty. I have never seen the batttery percentage above the 80% from the start. That's why i am connected so it always have power.