Blueforcer/awtrix3 #482: [FEATURE REQUEST] - Larger number for the clock display

wenzohaj, 2024-01-23:
# Feature Request

### Is your feature request related to a problem?

No - so only my eyes cause problems at a certain distance :-) 

### Describe the solution / feature you'd like

I have the clock in my living room and it is sometimes difficult to see the time. I have already chosen a color that is better visible. 

I have also deactivated display of the days of the week, so that in principle there would be more space for larger numbers. 

There would be three more lines available, but from a visual point of view, only two more usable lines in height would make sense. Here is an example of the actual situation and what it could look like:

[image]

Of course, this only works if the display of the weekdays is off, but as written, I have also deactivated this for me and it would then only be possible in this way.

Perhaps an option "Use large numbers" with the note that the days of the week cannot then be displayed would be possible.

Regards

Blueforcer, 2024-01-23:
There is only one font. At it's size is fix. No possiblity to scale it. 
This would need a whole new font, and is currently not planned

wenzohaj, 2024-01-23:
Ok, thank you for the quick answer!