Blueforcer/awtrix3 #483: [BUG] brightness issue with 3% and 4% and color #252525

leinardi, 2024-01-25:
# Bug report

## Describe the bug

There seems to be an issue with the color `#252525` and a brightness level between 3 and 4 percent. Interestingly the issue does not happen with a brightness of 2%

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.94

## To Reproduce

Steps to reproduce the behavior: 

1. set the calendar background and time color to `#252525`
2. set the brightness to 3%

## Expected behavior

The color should show at least at the same level of a brightness of 2% when the brightness is at 3-4%, instead the leds are completely off.

## Screenshots
2% brightness
[image]

3% brightness
[image]

## Additional context

Rebooting the device doesn't make any difference.

Blueforcer, 2024-01-25:
That's normal because the LEDs doesn't get enough power in low brightness to get the colors right.
For example the Blue LEDs draws more current than red or green and therefore turns off earlier.
Also the cheap LEDs are not calibrated from the manufacturer. 
Currently no fix for that

Blueforcer, 2024-01-25:
> That's normal because the LEDs doesn't get enough power in low brightness to get the colors right.
> Also the cheap LEDs are not calibrated from the manufacturer. 
> Currently no fix for that

leinardi, 2024-01-25:
But isn't strange that 2% works but 3% and 4% don't? Shouldn't 2% have the same issue?

Blueforcer, 2024-01-25:
> But isn't strange that 2% works but 3% and 4% don't? Shouldn't 2% have the same issue?

That's because of the gamma correction. It uses float values and push it to integer wich looses accuracy. 
Just try to use a slightly different color.

closed 2024-01-28 by Blueforcer