Blueforcer/awtrix3 #485: [BUG] Memory leak which overtime results in slow or no network response and unit becomes unresponsive

Bazz-Man, 2024-01-27:
# Bug report

Free memory slow drops resulting in a non responsive unit on the network although the time is still shown but with odd pixels flashing randomly.  Occassionally will auto reset and work until memory leaks away once again.

[image]

## Describe the bug

Monitoring the Free Ram via Home assistant it slowly reduces as per graph which occurs when sending messages to the indicator elements via HA's light on command in an automation. However if messages are sent via MQTT service in HA then no memory leak is seen even though it results in the same functional result (i,e, the indicator turns on).  The time taken for memory to leak away to impact the clock is directly related to the amount of messages sent.  i.e. if you send a message once per 30 seconds it will take around 15 hours to leak away but if you send it once every 3 seconds then it takes a few hours.  If the unit is reset/rebooted normal function is restored until the memory leaks away once more.  Whilst I appreciate this is not necessarily normal usage to send the same command at frequent intervals, I stumbled across it whilst doing some monitoring for another device and wanted a confirmed message being display via changing colours on the indicator but found the clock stopped responding after a few days, then through further testing found root cause.  I believe this could affect users over a number of days or even weeks with odd behavoiur and freezing only resolved through a restart.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - Also tested on a stand alone ESP32 TTGO OLED board to prove it was code and not hardware

## To Reproduce

Send via HA automation a turn on light command for the indicator

```
alias: XTEST - Awtrix memory test
description: ""
trigger:
  - platform: time_pattern
    seconds: /3
    enabled: true
condition: []
action:
  - service: light.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: light.awtrix_f07204_indicator_2
mode: single
```
This is sending an **ON** payload via MQTT to **awtrix_test/3c71bff07204/f07204_ind2/cmd_t**  topic

## Expected behavior

No memory loss should be seen as demonstrated by sending the command via MQTT service 

```
alias: XTEST - Awtrix memory test
description: ""
trigger:
  - platform: time_pattern
    seconds: /3
    enabled: true
condition: []
action:
  - service: light.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: light.awtrix_f07204_indicator_2
mode: single
```
This is sending a JSON **{"color":[255,0,0]}** payload via MQTT to **awtrix_test//indicator2**  topic

## Screenshots

Memory graph showing leak with messages being sent every 3 seconds via HA turn on light command
[image]

## Logs
None

## Additional context
None

Blueforcer, 2024-01-28:
It looks like an Issue with the used HA library. 
https://github.com/dawidchyrzynski/arduino-home-assistant
Since there have been no updates for two years, the problem will certainly not be fixed.

If youre using functions like this very often, i recommend to control it with mqtt. This also offers more functions.

closed 2024-01-28 by Blueforcer

Blueforcer, 2024-02-05:
> I have the same issues and my communtion is running only over mqtt

Unfortunately, I can't do anything with this information. Since thousands of other users do not have this problem, it would be important to know what exactly you are sending to awtrix.
The easiest way is to disable one automation after the other to isolate the json causing the problem.

HannesG79, 2024-02-29:
Hello, i changed the indicator on/off commands  from HomeAssistant (Discovered) to pure mqtt publish ... the memory leak is gone. Looks like the Homeassistant discovered switch is causing the issue.

Schermbecker, 2024-11-26:
> That's already the solution of this issue, but you told us you're already using pure mqtt commands?

It is obvious that the memory leaks when using the provided Awtrix 3 entities in Home Assistant (e.g. switch to "awtrix.next app").

Using plain mqtt topics does imho not leak the memory and works well.

Is there a chance that this issue will be fixed?

Blueforcer, 2024-11-26:
The used library for HomeAssistant discovery seems to be maintained for the time being. I don't think there will be a fix soon.

Schermbecker, 2024-11-26:
Thank you for the quick feedback.