Blueforcer/awtrix3 #486: Adding a custom App not working

EvilWatermelon, 2024-01-27:
I want to add a YouTube Channel subscriber count and add it via `/api/custom`. I'm getting a HTTP 200 but nothing happens even after a restart. Here's my code:

api.go
```go
const baseURL = "http://192.168.26.33/api/custom";

type apiPayload struct {
	Text		string `json:"Text"`
	Rainbow 	bool   `json:"rainbow"`
	Duration	int    `json:"duration"` 
	Icon		int    `json:"icon"`
}

func postToAPI(name string, p apiPayload) error {
	url := baseURL + "?name=" + name;
	
	jsonBytes, err := json.Marshal(p);
	if err != nil {
		return err;
	}
	
	resp, err := http.Post(url, "application/json", bytes.NewBuffer(jsonBytes));
	log.Println(resp);
	if err != nil {
		return err;
	}
	
	defer resp.Body.Close();
	
	if resp.StatusCode != http.StatusOK {
		return fmt.Errorf("%v", resp.StatusCode);
	}
	
	return nil;
}
```

youtube.go
```go
const ChannelID = "";
const ApiKey = "";

func youtubeStats() (uint64, uint64, error) {
	log.Println("YouTube Stats...");
	ctx := context.Background();
	service, err := youtube.NewService(ctx, option.WithAPIKey(ApiKey));
	resp, err := service.Channels.List([]string{"statistics"}).Id(ChannelID).Do();
	log.Println(resp.Items);
	
	if err != nil {
		log.Fatalf("%v", err);
	}
	
	if len(resp.Items) == 0 {
		log.Fatal("Channel not found");
	}
	
	stat := resp.Items[0].Statistics;
	return stat.SubscriberCount, stat.VideoCount, nil;
}
```

main.go
```
func main() {
	f, v, err := youtubeStats();
	if err != nil {
		panic(err)
	}
	
	log.Println(f, v);
	
	p := apiPayload{Text: fmt.Sprintf("%d/%d", f, v), Icon: 974, Duration: 4, Rainbow: true};
	err = postToAPI("youtube", p);
	log.Println(err);
	if err != nil {
		panic(err)
	}
}
```

Output:

```
2024/01/27 16:15:56 YouTube Stats...
2024/01/27 16:15:56 &{200 OK 200 HTTP/1.1 1 1 map[Access-Control-Allow-Headers:[*] Access-Control-Allow-Methods:[*] Access-Control-Allow-Origin:[*] Content-Length:[2] Content-Type:[text/plain]] 0xc0003965c0 2 [] true false map[] 0xc000454000 <nil>}
2024/01/27 16:15:56 <nil>
```

Blueforcer, 2024-01-27:
I don't know Go,

Please post the raw JSON wich is generated and sent to awtrix.

EvilWatermelon, 2024-01-27:
```json
{
      "text":"3570/62",
      "rainbow":true,
      "duration":4,
      "icon":974,
      "save":true}
```

EvilWatermelon, 2024-01-27:
It worked but without the Icon.

10der, 2024-01-27:
> It worked after removing the USB from my PC and restarting the box but there's no icon.
Did u upload 974 icon to device?

Blueforcer, 2024-01-28:
Good question...

EvilWatermelon, 2024-02-06:
> > It worked after removing the USB from my PC and restarting the box but there's no icon.
> 
> Did u upload 974 icon to device?

Yes

Blueforcer, 2024-02-06:
> > 
> > Did u upload 974 icon to device?
> 
> Yes

How did you upload it?
If manually, to which folder?
please upload this icon here

EvilWatermelon, 2024-02-06:
I uploaded it manually to the ICONS folder

Blueforcer, 2024-02-06:
can you please upload the icon here in github?

closed 2024-03-06 by Blueforcer