Blueforcer/awtrix3 #487: How to send an empty MQTT message to delete a custom app

Jackie78De, 2024-01-30:
Hello,

I am currently having trouble finding the right syntax for dfeleting a custom app I have added. In the documentation it says that I need to publish an empty MQTT message to do so. But I fail to dind the right syntax.

In my client, something like this works to set an app:

set MQTT2_FHEM_Server publish awtrix_123456/custom/PowerPlugOn {"text" : "on"}

But whats the correct syntax to delete it? I tried
set MQTT2_FHEM_Server publish awtrix_123456/custom/PowerPlugOn {"text" : ""}

and
set MQTT2_FHEM_Server publish awtrix_123456/custom/PowerPlugOn {""}

But none of them seem to work. Any help would be highly appreciated. Thank you!

Blueforcer, 2024-01-30:
Try {}

85Kid, 2025-01-25:
I have a similar issue. The deleted app appears again after a reboot.

Blueforcer, 2025-01-25:
> I have a similar issue. The deleted app appears again after a reboot.

Youve sent the MQTT message as retained, wich means the MQTT broker sends the message on every connect.

85Kid, 2025-01-25:
How can I change that? Sorry, I’m not that experienced with MQTT

Blueforcer, 2025-01-25:
Send a empty payload WITH retained to the same topic.

PeterSpie, 2025-03-26:
Hi, sorry to ask an other question about this: I have send an empty message, and now the screen is just black for the duration of the empty message. How can I completely delete it?