Blueforcer/awtrix3 #488: [Configure First day of week]

KarstenDE, 2024-01-31:
# Feature Request

### Is your feature request related to a problem?

In the clock view, the day indicator (lower two leds per day) starts showing we weekday with Sunday. In Germany and some other countries this counting is "wrong".

### Describe the solution / feature you'd like

Make the first day of the week configurable and use this in the clock app.
E.g. in Outlook you configure this and the calendar view use this.

### Describe alternatives you've considered

The configuration can happen in the web ui or via MQTT.

### Additional context

N/a

Blueforcer, 2024-01-31:
Das lässt sich mit der settings API umstellen. Standardmäßig ist der Montag bereits der Anfang der Woche.

KarstenDE, 2024-01-31:
Cool, Thanks! Is this documented somewhere? I did not find it in the docs ...

Blueforcer, 2024-01-31:
https://blueforcer.github.io/awtrix-light/#/api?id=json-properties-1

Key "SOM"