Blueforcer/awtrix3 #489: [BUG / FEATURE REQUEST] MQTTS (MQTT via TLS) is rejected

Moltey, 2024-02-03:
# Bug report

## Describe the bug

- I want to connect to a MQTT server via TLS connection (MQTTS). 
- The TLS Certificate is valid (Let's Encrypt Certificate). 
- I double checked MQTT Server, Username & Password.

However the connection is refused by the MQTT Server (Mosquitto):

`OpenSSL Error[0]: error:1402542E:SSL routines:ACCEPT_SR_CLNT_HELLO:tlsv1 alert protocol version`

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001) [brand new]
  - AWTRIX-LIGHT version: v0.94

## To Reproduce

Try to establish a MQTTS connection.

## Expected behavior

MQTTS connection is expected to be established.

## Screenshots

n/a

## Logs

`OpenSSL Error[0]: error:1402542E:SSL routines:ACCEPT_SR_CLNT_HELLO:tlsv1 alert protocol version`

## Additional context

n/a

Blueforcer, 2024-02-03:
TLS is currently not supported.

Moltey, 2024-02-03:
Thank you for your feedback. I assumed, I configured something wrong.

Side note: I am using [OpenDTU](https://github.com/tbnobody/OpenDTU) that provides MQTTS (MQTT TLS) on an ESP32, so it might be possible to implement. However, as I never coded anything on ESP32 myself, I would have no idea where to start.

Blueforcer, 2024-02-06:
It order to establish TLS connection with the mqtt server a CA certificate has to be defined.
This and the SSL handshake costs huge ammount of RAM, wich forces AWTRIX to crash because it getting out of memory.
So unfortunally this is currently not possible with awtrix. sorry

niklaspeura, 2026-02-03:
I'm also missing this feature, all my other components communicate over TLS using my own CA.