Blueforcer/awtrix3 #491: [BUG]Custom APP with variable pagecount displays non existing pages

AndiGloor, 2024-02-05:
# Bug report

## Describe the bug
I h've designed an custom app showing my PV data. During daylight it contains 3 pages, whereas after sunset only 2 pages are neccessary. It works fine going from 2 pages to 3 pages at sunrise. But at sunset, when the app only contains 2 pages, it displays still the 3rd page from the daylight-period. After sending the clock to sleep it works perfectly fine again and displays the 2 page version.

## Additional information
I publish the app via MQTT, using Node-Red.
- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.93

### App with 3 pages:
`[{"text":"0.5kW","icon":21256,"noScroll":true,"textCase":2,"duration":10},{"text":"0.1kW","icon":52452,"noScroll":true,"textCase":2,"duration":10},{"text":"62.9%","icon":32057,"noScroll":true,"textCase":2,"duration":10}]
`

### App with 2 pages:
`[{"text":"0.5kW","icon":21256,"noScroll":true,"textCase":2,"duration":10},{"text":"62.9%","icon":32057,"noScroll":true,"textCase":2,"duration":10}]
`
## To Reproduce
Publish the app with 3 pages via MQTT. Then republish the 2 pages version.

## Expected behavior
The clock shoud always show the app as published, displaying as many pages as contained in the array.
Or the api documentation shoud provide instructions on how to deal with that case ;-).

Blueforcer, 2024-02-05:
I think there is a misunderstanding. Youre just sending 3 apps at once.
Please read 
"Handling of Multiple Custom Pages" in the documentation.

I will rename this part from pages to apps.

AndiGloor, 2024-02-05:
Thanks for the clarification.

So to archive my goal I have to change my implementation to publish on 3 topics simultaniously and "unpublish" (publish empty topic) when reducing the pagecount? 

My first implementation used different topics, but the downsinde was that the clock switched to the newest published app immediately. Now I pubish updates after the summarized duration of every page (when something changes). This worked well, appart from the "page reduction issue". 
Is there a way to avoid the "display immediately" behaviour when publishing an app update? This would simplify things a lot.

AndiGloor, 2024-02-05:
I ment:
> Updating: To refresh a custom page, dispatch a modified JSON object to the identical endpoint. The display updates instantly.
The clock immediately changes to the "refreshed" custom page. 

Bit now I'v realised, that I should use 2-3 different topics and send an empty payload if I like to remove a app/page.

Thanks a lot!

closed 2024-02-05 by AndiGloor