Blueforcer/awtrix3 #498: [BUG] temp_offset leads to wrong temperature value

individualist99, 2024-02-22:
# Bug report

## Describe the bug

My ulanzi clock shows a temperatre that's 1°C lower than it should bei. So I added an temp_offset of 1 to dev.json and rebooted the clock. It now Shows 30°C instead of previously 20 °C / expected 21°C   
  
Looks Like temp_offset does Not Work as expected. 
  
content of dev.json:

```
{

  "temp_offset": 1,

  "hum_offset": 0
}

```## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.93

## To Reproduce

Set temp_offset to 1 
## Expected behavior

Temperature should be 1°C higer than before
## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

## Additional context

Add any other context about the problem here.

Blueforcer, 2024-02-22:
Standard value is 9, because in reality the raw sensor value is completely wrong. 

So you need to enter 8

closed 2024-02-26 by Blueforcer

Blueforcer, 2024-05-08:
Make sure there is no leading space in your JSON. The editor will show you the error with a red dot.

proffalken, 2024-05-08:
Nope, no red dots:
[image]

Blueforcer, 2024-05-08:
Did you restarted awtrix after change?
For me it's working.

proffalken, 2024-05-08:
I'll restart it again, but I thought I had.

Does it need to be a full power-off/back on, or will "restart ESP" from the MQTT page do the trick?

proffalken, 2024-05-10:
@Blueforcer - I've edited the file, saved it, but when I go back in, the save hasn't taken effect.

This remains the same regardless of whether I reboot or not, the "temp_offset" value remains at `-9`