Blueforcer/awtrix3 #501: [BUG] Web Interface Does Not Work on a Different Port

m33x, 2024-02-26:
# Bug report

## Describe the bug

If AWTRIX Light is not served on port 80 (HTTP) or port 443 (HTTPS), the web interface does not work as certain resources like `DoNotTouch.json` cannot be reached.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ v0.94 ]

## To Reproduce

Steps to reproduce the behavior:

1. Serve AWTRIX Light on a different port than 80
```ssh home -L 127.0.0.1:1338:192.168.2.3:80```

2. Visit the web interface of AWTRIX Light in your browser
```firefox http://localhost:1338```

## Expected behavior

The web interface loads as expected.

## Screenshots

[image]

## Logs

```text
GET http://localhost/DoNotTouch.json net::ERR_CONNECTION_REFUSED
(index):140 Fetch failed loading: GET "http://localhost/DoNotTouch.json".
```

## Additional context
Affected are:
```JavaScript
var e = new URL(`${window.location.protocol}//${window.location.hostname}/scan`);
var e = new URL(`${window.location.protocol}//${window.location.hostname}/status`);
var e = new URL(`${window.location.protocol}//${window.location.hostname}/DoNotTouch.json`);
var e = new URL(`${window.location.protocol}//${window.location.hostname}/restart`);
```
Potential fix is using [window.location.origin](https://developer.mozilla.org/en-US/docs/Web/API/Location/origin)
```JavaScript
> window.location
host: "localhost:1338"
hostname: "localhost"
href: "http://localhost:1338/"
origin: "http://localhost:1338"
pathname: "/"
port: "1338"
protocol: "http:"
```
```JavaScript
> var e = new URL(`${window.location.origin}/DoNotTouch.json`);
> e.href
'http://localhost:1338/DoNotTouch.json'
```
Is the current code potentially exploitable and a XSS vulnerability? [Unsure](https://security.stackexchange.com/search?q=+window.location.hostname), but likely not.

m33x, 2024-02-26:
So this is a won't fix, I assume? Would you merge it, if I create a PR?

Blueforcer, 2024-02-27:
> So this is a won't fix, I assume? Would you merge it, if I create a PR?

PR for what?
Using window.location.origin again would undo the other PR wich helps a lot of people.

m33x, 2024-02-27:
A PR with window.location.**origin**, this fixes my issues, and does not break anything for people using window.location.**protocol** **and** window.location.**hostname**. If at all, it makes their version even better and more complete.

m33x, 2024-03-12:
An option to change the port of the Webinterface unfortunately doesn't fix the issue.

m33x, 2024-03-12:
I have tested the new version. This fixes the problem. Thanks!

closed 2024-03-12 by m33x

Blueforcer, 2024-03-12:
Thanks goes to you!