Blueforcer/awtrix3 #503: [BUG] Online Flasher

pesce, 2024-02-28:
# Bug report

website: https://blueforcer.github.io/awtrix-light/#/flasher
1. Using multiple windows 10 computers, Chrome and Edge browsers, I cannot get the flasher to appear.  
2. On the same page, when I click on the ESP Web Tools link (which opens in the same page) and attempt their live demo, I get this error: Error: Failed to execute 'requestPort' on 'Serial': Access to the feature "serial" is disallowed by permissions policy.

When I go directly to https://esphome.github.io/esp-web-tools/ and attempt the same live demo, it works as expected.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

## To Reproduce

Browse to: https://blueforcer.github.io/awtrix-light/#/flasher

## Expected behavior

I expect a "connect" button, or similar.

## Screenshots

If applicable, add screenshots to help explain your problem.

[image]

[image]

[image]

## Additional context

Thank you

Anonym-tsk, 2024-02-28:
Go to chrome://settings/content/siteDetails?site=https%3A%2F%2Fblueforcer.github.io
[image]
Look at "Serial ports" permission. Set it to default value.

Blueforcer, 2024-02-28:
mhh. still works as expected:
[image]

closed 2024-02-28 by pesce