Blueforcer/awtrix3 #505: [FEATURE REQUEST] Disable app switch via hardware buttons

Green-Kite, 2024-03-02:
# Feature Request

### Is your feature request related to a problem?

No.

### Describe the solution / feature you'd like

I would like to disable the app switch via the buttons on top of the Ulanzi Clock. In that way I could use the Buttons as custom triggers for Home Assistant Automations, like an alarm Clock or other things based on button presses without leaving the app I set via MQTT.

### Describe alternatives you've considered

This could be added as an option to the dev.json.

Blueforcer, 2024-03-02:
Already possible.
See BLOCKN key in the settings API

Green-Kite, 2024-03-03:
I'm sorry, I read the documentation but apparently overlooked this line.

closed 2024-03-03 by Green-Kite