Blueforcer/awtrix3 #510: Disable Battery Display within dev.json

fidoriel, 2024-03-13:
It would be great to disable the battery level without mqtt with a simple setting in `dev.json`. So that sending `[{"name":"bat","show":false}]` to `/app` is not necessary anymore.

MrHollowPS, 2024-03-13:
You can disable the battery thing from within the clock settings. hold the middle button on the clock then navigate until you reach the Apps menu and in there battery off.

Blueforcer, 2024-03-13:
> It would be great to disable the battery level without mqtt with a simple setting in `dev.json`. So that sending `[{"name":"bat","show":false}]` to `/app` is not necessary anymore.
> 

this API doesn't exist in the documentation, and shouldn't be used.

you can use the official App or API to turn it off permanently

fidoriel, 2024-03-13:
> You can disable the battery thing from within the clock settings. hold the middle button on the clock then navigate until you reach the Apps menu and in there battery off.

Why is this not documented?

fidoriel, 2024-03-13:
Did not expect to find it there :)

closed 2024-03-13 by fidoriel