Blueforcer/awtrix3 #512: [BUG] Not reaction for rainbow text option in MQTT service

zerofull56, 2024-03-15:
# Bug report

## Describe the bug

If I add a line with "rainbow":"false" to the script in HA for send a message to the clock, the text will be colored.

## Additional information

- Devices involved:
  - Model: Awtrix 2
  - awtrix3 version: 0.95
## To Reproduce

Steps to reproduce the behavior:
Send for MQTT service in HA:
"text":"Hello!",
"rainbow":"false"

Blueforcer, 2024-03-15:
You need to send it as Boolean value not as string

"rainbow":false

zerofull56, 2024-03-15:
Oops, it's my fault. I using template for this script, and input_boolean.yml for "rainbow", who send "on" and "off" instead "true" and "false"... May be i need use some soft switch instead boolean. Thanks for explanation, now, can close the issue.

closed 2024-03-15 by zerofull56