Blueforcer/awtrix3 #513: [QUESTION] What do red-blue-green indicator mean ?

rhuss, 2024-03-16:
I'm sorry to "misuse" issues for asking a question for which I couldn't find an answer yet, neither on the web nor in the documentation (apologies in advance).

From time to time, suddenly, three indicators appear that I have not programmed and that only vanish when I reboot:

[image]

I see this simultaneously on two Ulanzi devices simultaneously (i.e., they appear simultaneously). I'm using Awtrix3 v0.96 on both devices (but this has already happened on older devices). Also, I have a password set on the devices, so I don't think indicators have been set via the Rest API from the outside.

My questions:

* Is this part of Awtrix3 core functionality ?
* If so, what does it indicate, and how can it be eliminated?
* If not, do you have any suggestions for debugging this issue?

thanks! And thanks for this great piece of software; I love it!

Blueforcer, 2024-03-16:
Yes this is a functionality of awtrix called indicators.

https://blueforcer.github.io/awtrix3/#/api?id=colored-indicators

If you're using Homeassistant, this indicators are handled as lights. So if you have a automation wich turn on all lights, the indicators will also turn on.

rhuss, 2024-03-16:
Yes, I'm using indicators for my custom handling, but I'm calling Awtrix only from NodeRed, and I don't have any integration that switches on all three indicators.

I have Home Assistant installed, and indeed all indicators are available as switches. And HA has recorded the switch-on event, but according to the log, it hasn't switched it on itself.

[image]

However, I just wanted to be sure that switching on the indicators in red-blue-green is not a core feature of Awtrix itself to indicate a particular state (like that an update is available), so I can continue hunting elsewhere.

Thanks!

rhuss, 2024-03-16:
Ah, I found it:

[image]

@Blueforcer you were absolutely right, an 'switch-all on' action triggered the indicators (i was not aware that of that action in Alexa).

Thanks again!

closed 2024-03-16 by rhuss