Blueforcer/awtrix3 #514: [BUG] unauthorized error - Icon Designer

hugihug, 2024-03-16:
# Bug report

## Describe the bug
It ist not possible to save an created icon to the AWTRIX device if an user and password is set.

## Additional information
Dialog Error unauthorized appears

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

Blueforcer, 2024-03-16:
Please be more detailed

hugihug, 2024-03-16:
I designed an icon with the icon editor and try to save it with "Save to AWTRIX" after entering a name.
This results in Dialog box which says ERROR UNAUTHORIZED.
 In the device a user and a password is set. So I assumed that there is an issue with the authorisation.

Blueforcer, 2024-03-16:
App update is in preview.

closed 2024-03-17 by Blueforcer