Blueforcer/awtrix3 #516: [FEATURE REQUEST] Switch on on external power restore / power off when external power is disconnected

HochdruckHummer, 2024-03-23:
I'm using my Ulanzi/Awtrix devices in my dedicated nerd-room / mancave. The room has a separate powerswitch to shut down everything when I leave. Background: There are HF radios and antennas, so it's important for me to switch everything of in case of a lightning strike to the antennas and resulting electric surge.

When I come back and power is restored, the clocks are powered off - because the battery died in the meantime of course.
So I have to power them back on manually by pressing the two arrow buttons.

Request: When the clock gets electric power again, it should automatically come back to life. 

Furtehr idea: It would also be nice if one could deactivate the usage of battery power and use external power supply only. Would save battery lifetime in my case, as the battery will run low over night everytime I leave the room. So a switch to say "power off when external power gone" would be very useful.

Blueforcer, 2024-03-23:
Neither is possible in terms of hardware. The ESP32 has no communication to the power management

HochdruckHummer, 2024-03-23:
Thank you for the fast reply! Have a nice weekend!

closed 2024-03-23 by HochdruckHummer